PubInv/NASA-COG #30: Add a timer "watchdog" to react to software hangs...

RobertLRead, 2022-09-08:
Create a timer/ISR which performs an emergency shutdown if superloop times out.

LokiMetaSmith, 2023-07-19:
This issue will need to be coordinated with AmOx, as we will need a "always safe to do X" strategy that may differ from device to device

ForrestErickson, 2023-09-03:
Is there more to capture now in this issue to reflect the current status?

LokiMetaSmith, 2023-09-03:
This is related to issue #188 E - stop, which would be an external trigger for this. 

It's been confirmed that "safe shutdown" strategy is to remove power and blower fan from the stack, in almost every case, so that is very easy to implement in hardware with enable and Power OK signals, as well as disabling power to power supplies, heater level logic and fan controller hardware. 

We will need a state transition table to reason about all of the various things and if the logic needed to drive them is positive or negative, and ideally we can relate this as a low level system state that can be represented as a visually observable LED array indicator or similar. 

This should be put together as part of the PCB layout for Control Board v2.

ForrestErickson, 2023-09-05:
Vocabulary. 
I think we have an opportunity to confuse ourselves.

Today, 20230904 I wrote some software which enables the Due watch dog dimmer which will reset the Due if the main loop where the dog is fed takes too long.  In a well designed system the Due will boot up, report the watch dog time out event and resume controlling to the best of it's ability.
This seams best described as a controller watch dog reset.

The failures above about which we speculated which not going to be fixed by a controller reset require a counter measure more basic which for example perhaps opens the AC input (by a crow bar for example) 
This needs a distinct name which can not be confused with the controller watchdog.

I like **crow bar** but want other suggestions.

RobertLRead, 2023-10-09:
It think is closable, except that the toggle of the external GPIO pin is a good idea which has not been done. I am assigning this to the next milestone for that to be accomplished.

ForrestErickson, 2023-10-09:
I believe we should conduct a failure mode effects analysis, FMEA, for the current design and even if we do not change Serial number 1 and  or 2 we will develop a more mature plan going forward.

### FMEA Questions
What happens for each wire connecting components is open or becomes open?
What about some shorts to adjacent wires or to power or ground?
What about power drop outs of arbitrary duration?
Other?
 
### Benefits
The results of this analysis might change our design.
The results of this analysis might inform our development of "factory test" which is applied to every unit before it is deemed functional for release to the field, ( aka sellable)