PubInv/NASA-COG #40: Create an absolute limit on the rate of the current change to the stack.

RobertLRead, 2022-09-11:
We need to put an absolute limit on the rate of the current change to

LokiMetaSmith, 2023-07-12:
this should also incorporate a thermal delta rate change as well. 
In operation this would be a PID control loop with guard rails to protect against overshooting that ties together thermal rate limits, current rate limits, and heater rate limits all in accordance to their contribution to the total balance of thermal contribution and delay in thermal flux transfer to the algorithm.

ForrestErickson, 2023-09-03:
@RobertLRead ,
Status?
Can this be closed or marked Will Not Fix?

ForrestErickson, 2023-09-18:
@RobertLRead 
Has this branch become moot?

RobertLRead, 2023-09-18:
The absolute temperature change is handled; this issue could be closed
because of that.

However, I do allow the amperage to change arbitrarily; but under the "5
knob" decision this is to be expected.

I suggest we close this.

RobertLRead, 2023-11-17:
This is obsolete in the new algorithm, I think.

closed 2023-11-17 by RobertLRead