PubInv/NASA-COG #46: Can't upload when 12V plugged into Due...

RobertLRead, 2023-03-02:
The work around is to plug 12 V power in after it is running.

I don't understand this, but clearly it is a question of the draw on the 12V power some how. This could be solved by not taking the 12V power from the VIN pin of the Due, probably.

RobertLRead, 2023-03-11:
Our current breadboard has a significant problem.

When the Motor driver and at least one PWM pin is plugged in and 12V is supplied to the fans, the Due won’t reset. You can turn it off after the Due is up, but you can’t reset in this situation.

An oscilloscope on the IOReset pin shows that it gets pulled to 6 volts and stays there. However, why this is occurring is unknown.

I have placed a large cap on 12V power to ground; that does not seem to be the problem. I conject that power is being pulled from the PWM pin, but since these should be low anyway, I am not sure that makes sense.

These fans have given us no end of trouble. This is a low priority (it has a work-around), but we need to do something and we need to control this.  This is the messiest part of the my breadboard anyway. 

Candidate actions are:
Create a PCB for this section of the bread board
Isolate and test separately
Ignore.

RobertLRead, 2023-03-11:
A fundamental problem here is that the PWM wires are held at 5-7 Volts when the power supply is 10 Volts.  I have no idea why this should be so, but it is entirely unsafe to attach 7V to in output pin of a DUE! Possibly These values are floating, but they don't seem to be floating---it appears that either the fans that we purchased or my small attempts to solder them together has caused them to behave in flagrantly bad manner. I am unsure what to do about this.

LokiMetaSmith, 2023-07-12:
I'm assuming per #81 and related issues, that this is no longer a problem.

ForrestErickson, 2023-09-03:
@RobertLRead ,
Has this been resolved?
Can this issue be marked done and closed?

CC: @LokiMetaSmith

closed 2023-10-06 by LokiMetaSmith