PubInv/NASA-COG #47: Creating a slight positive bias on the stacks.

RobertLRead, 2023-03-02:
We have been told that stacks should always have a slight positive bias, like 0.1 Volts. The most reliable way to provide this is to design circuit for it; I assume we could use a voltage divider circuit from a 12V source and a diode to accomplish this.

LokiMetaSmith, 2023-05-18:
This is only an issue if the cables for the power supply can be reversed. There should be no need for a biasing circuit, as a DC power supply should provide sufficient biasing under nominal operational control.

LokiMetaSmith, 2023-07-12:
Per the last NASA-COG meeting, we should revisit this requirement. 
In the meeting they mentioned having value in being able to swap polarity to purge the stack of contaminants. I imagine that the bias voltage is similar. the primary issue is that when the unit is "parked", the voltage on the power supply is indeterminant. 
We will need to measure the ESR of the system to determine the amount of voltage and current needed to maintain such a polarization and for how long. 

I suspect that we will need to introduce more system states, such as "Cold Start" and "Warm Start" such that a cold start assumes that the stack is in an unknown configuration due to being in storage, a hard reboot of the system, or a lack in continuity of the system power, where as a warm start, has maintained a constant system control loop over the entire operational up time of the system.

ForrestErickson, 2023-09-03:
IIRC I have heard @RobertLRead  report that a bias is implemented in the code during August project meetings.

Can this be closed or is further investigation warranted?

What is the appropriate milestone for this?

RobertLRead, 2023-10-09:
This is now a duplicate of #306 based on our newer understanding; I am closing in favor of that.

closed 2023-10-09 by RobertLRead