PubInv/NASA-COG #48: Create a special task for testing the fans independent of the rest of the system...

RobertLRead, 2023-03-02:
Our Sanyo fans are finicky. These are 4-wire fans with a tachomter and are in theory controllable by PWM (they spec calls for a 25KHz pwm signal, and a Due doesn't produce that out of the box.) We have 4 electical fans mechanically treated as a single unit. Because there is a lot of wire for the PWM and the tachometer to be read, we need a separate PlatformIO configuration mode that just tests that fans.

LokiMetaSmith, 2023-07-12:
This should control fan speed separately from the system for test purposes.
https://www.amazon.com/Taidacent-2510-4P-Controller-Control-Connector/dp/B0BHNC776L/

ForrestErickson, 2023-09-03:
@RobertLRead ,

Is this now moot and can be closed?

LokiMetaSmith, 2023-09-03:
I believe so, but need confirmation

RobertLRead, 2023-09-03:
I'm closing this as no long relevant.

closed 2023-09-03 by RobertLRead