PubInv/NASA-COG #64: The front panel on the Oxygen Engine Digital Control System should have USB ports...

RobertLRead, 2023-04-18:
We should probably have ports for both the Arduino Due (one micro-usb) and also whatever ports a modern Raspberry PI needs for serial communication, keyboard, and monitor.

LokiMetaSmith, 2023-05-18:
It'll be easier to design with short pigtail connectors, rather than trying to panel mount either the arduino or rPi/single board computer. 

Google searches for "hdmi/USB panel mount cable" have several ready made options.

ForrestErickson, 2023-09-03:
@LokiMetaSmith 
Can you document the solution for Serial number 2 as implemented here?

LokiMetaSmith, 2023-09-03:
The usb and Ethernet ports were moved to the back panel as part of the
missing back panel order.

In doing so, I used a 30mm diameter hole, and that turned out to be too
large for the adafruit usb and ethernet plugs. I have since made washer
adapters to expand the holes so they will now fit.

ODECS unit 1 does not have these holes, and so properly sized holes should
be drilled into the back plate for usb and ethernet.

LokiMetaSmith, 2023-10-06:
Future versions of the front panel should have the holes on the front as it is currently envisioned that the cfc will be a self contained system.

closed 2023-10-06 by LokiMetaSmith

ForrestErickson, 2023-10-06:
Regarding, "The usb and Ethernet ports were moved to the back panel as part of the
missing back panel order."

Are their photos of Serial number 2 as implemented?

Can such photos please be added at:  https://github.com/PubInv/NASA-COG/blob/develop/mech/README.md ?