PubInv/NASA-COG #65: Choose Optimum Color (And Interlock) For COG Current Connection (Anderson Connector)

ForrestErickson, 2023-04-18:
When the typical voltage for the COG becomes known we should consider the color scheme that the Wikipedia page references as Anderson Suggested.
If I had seen this page before ordering from Powerwerx I would have chosen yellow.
> [image]

Before we order materials for additional power channels we should consider the color and order enough to replace the existing cables plastic shells too.

Lee

LokiMetaSmith, 2023-05-18:
I would note that Anderson only suggests the color coding and it is obvious that in practice that the suggestion is ignored for the wide array of usage cases. Case in point, I didn't even know there was a standard because all of the devices I have experience with, namely APC battery packs use a combination of Grey and Yellow anderson connectors. 

Therefore, I would suggest using Orange, as it's not commonly used yet, it fits our voltage range of 12-15v and we should label the cables, machine nameplate or ports with the actual voltage supplied by the anderson connectors.

Or define this as a non-issue/won't fix

ForrestErickson, 2023-09-03:
I think I can summarize this status for the mile stone of Serial number 2 as will not fix.

With the delivery of Serial number 1 to the user changing the connector type may cause confusion.

We might be best off closing this as for ever will not fix and continuing with the commonly available grey interlock shell.

LokiMetaSmith, 2023-10-06:
agreed

closed 2023-10-06 by LokiMetaSmith