PubInv/NASA-COG #70: Can We Detect Fan Obstruction With Tachometer?

ForrestErickson, 2023-05-18:
# Can We Detect Fan Obstruction With Tachometer?

To answer this question, I have made experiments on measuring the RPM on a fan with and with out an obstruction. I did not find significant signal in the RPM to from which to infer obstruction.

## Test Results
This is a plot of fan RPM with the fan set at a PWM of about 50%. The first half of the plot I had the fan obstructed with a piece of paper choking off most of the air flow. Mid way through the plot I removed the obstruction.
[image]

Same experiment but with the fan set for PWM to maximum.
[image]

## Test Description
Using a 92x92mm fan Foxcomm part number PV902512PDPF 0H
Powered from 12V.
Used Arduino UNO sketch VERSION "Rev: 0.6"

Lee Erickson

ForrestErickson, 2023-05-18:
Link to tagged firmware: https://github.com/ForrestErickson/fanRPM/tags  Use Tag to version 0.6

RobertLRead, 2023-10-09:
I would like to see this repeated within the CFC with our actual hardware, so I am leaving it open.