PubInv/NASA-COG #72: Power Supply Briefly comes on full-power before being initialized....

RobertLRead, 2023-05-24:
The digital DC power supply that we use comes on full-power when first connected before the digital system connects to this. This is not noticable when the load has high resistance, but when the load is low-resistance, this can pull a large number of amps and power, which potentially can heat the load dangerously. 

We need to research how to set this to not apply voltage until commanded. Possibly this can be done with just grounding something.

LokiMetaSmith, 2023-06-01:
there should be an enable pin on the power supply, this likley will need to be pulled either high or low and should act as a safety disable until the Due has the device under control.

ForrestErickson, 2023-09-03:
@LokiMetaSmith ,
@RobertLRead ,

Regarding, "there should be an enable pin on the power supply, "

I am finding I wish I had reviewed these issues before release of the Control V1 PCB.
A quick check of the power supply pins shows there is an enable AKA inhibit pin on the 9th pin and 11th pin.

On the plus side our Control V1 PCB has a header, J14 which breaks these pins out for easy access. 
I have not read about how these pins are used. I assume we need pull down or pull up resistors to force the state till the Due can come to life.

Pin table for power supply below.

Screen shot showing J 14 location on Controller V1 PCB
[image]

Power supply header Pins and Signals

| **Pin #** 	| **Function** 	| **Description**                      	| **Pin #** 	| **Function** 	| **Description**                      	|
|-----------	|--------------	|--------------------------------------	|-----------	|--------------	|--------------------------------------	|
| 1         	| VS+          	| Remote Sense (+)                     	| 13        	| ACI          	| I Program                            	|
| 2         	| VO+          	| Positive Output Voltage              	| 14        	| GND          	| Ground                               	|
| 3         	| VS-          	| Remote Sense (-)                     	| 15        	| VCI          	| V Program                            	|
| 4         	| VO-          	| Negative Output Voltage              	| 16        	| GND          	| Ground                               	|
| 5         	| POK          	| Power OK                             	| 17        	| AUX          	| +5V/0.5A or +9V/0.3A Auxiliary Power 	|
| 6         	| GND          	| Ground                               	| 18        	| GND          	| Ground                               	|
| 7         	| PAR          	| Parallel operation current share     	| 19        	| SCL          	| Serial Clock I2C                     	|
| 8         	| VSET         	| Aux outpu setting                    	| 20        	| SDA          	| Serial Data I2C                      	|
| 9         	| EN-          	| Inhibit ON/OFF(-)                    	| 21        	| GND          	| Ground                               	|
| 10        	| GND          	| Ground                               	| 22        	| GND          	| Ground                               	|
| 11        	| EN+          	| Inhibit ON/OFF (+)                   	| 23        	| RX           	| For RS232 Receiver function          	|
| 12        	| AUX          	| +vV/0.5A or +9V/0.3A Auxiliary Power 	| 24        	| TX           	| For RS232 Transmission function      	|
|           	|              	|                                      	|           	|              	|                                      	|

LokiMetaSmith, 2023-10-06:
This should be left for Control v2 changes

ForrestErickson, 2023-10-06:
Can " ... we need pull down or pull up resistors to force the state till the Due can come to life." this be tested with some of our real supplies and Control v1 boards?  Solder something to the header pins next to the connector for example?

LokiMetaSmith, 2023-10-09:
correct

closed 2023-10-09 by LokiMetaSmith

ForrestErickson, 2023-10-11:
@RobertLRead 
@LokiMetaSmith 

## Due Enable of Power Supply Outputs
On Control V1.1 PCB I have wired up control of the power supply enable output under the Due control.
The circuit is a MOSFET transistor which when the gate is made high the power supply will be enabled (each separately). When the Due is in reset and before the program takes control the transistor is off and the power supply outputs will be disabled.

PS1_EN is on D23
PS2_EN is on D8

[image]