PubInv/NASA-COG #73: Add a complete component list/BOM to the report

RobertLRead, 2023-05-25:


ForrestErickson, 2023-05-26:
@RobertLRead 

20230526 7 AM Status:
[Lee has been updating the README.md for the Front Panel which as a parts list.](https://github.com/PubInv/NASA-COG/tree/develop/mech/frontpanel)
[Lee has created README.md for the Enclosure which as a parts list.](https://github.com/PubInv/NASA-COG/tree/develop/mech/enclosure)

Both have a mark down table  with fields:
<html>
<body>

Quantity | Description | Manufacturer | Man Part Number | Distributor | Distributor Part Number | Drawing | Cost Each
-- | -- | -- | -- | -- | -- | -- | --

</body>
</html>

Soon but not yet integrated into the google document:
_An Oxygen Engine Digital Control System (DRAFT v.0.1)_
https://docs.google.com/document/d/1MtXPKEr9y4HvgYWzUUQt9Jl5vSoTlWfLfREPRGYORko/edit#heading=h.ko2ebw87ndbn

ForrestErickson, 2023-05-26:
@RobertLRead 

20230526 about 1:00 PM Eastern Lee has pasted the tables for the Front Panel and the Enclosure into the document referenced above.

closed 2023-11-17 by RobertLRead