PubInv/NASA-COG #78: Control of 24 Volt Fan with Motor Control by PWM Not Working Well

ForrestErickson, 2023-05-27:
@RobertLRead reports that his effort to control the 24V Fan Make ??? and PN ???
with a motor controller board and by PWM is not working well.
With the PWM duty cycle set anywhere but near 100 % the fan does not work.

Fan make: 
Fan Model:
Motor Control Board:

Firmware:

ForrestErickson, 2023-05-27:
Lee is attempting to duplicate Roberts fan problem.
Lee has used a both a 92mm Foxcomm fan  and a 120mm  GDSTIME GDP1238 120mm fan as test fans.
Lee has used L298N Motor Driver Controller Board Module https://www.amazon.com/HiLetgo-Controller-Stepper-H-Bridge-Mega2560/dp/B07BK1QL5T/ref=sr_1_1_sspa wired to make a PWM of the +12 to the fans.

Connection description.  Connect the fan +12V to the output 1 of the motor control board. 
Under control of an UNO with frequency of 976 Hz measured PWM the fans make unpleasant acoustic noise.
They do control for speed.
The FG (Tachometer signal) is modulated with the PWM .

[image]

Tentative conclusion. Using a motor control to PWM the raw power into the fan is not a good idea. 
Not sure if I have duplicated the problem with @RobertLRead 's 24V fan or not.

RobertLRead, 2023-05-30:
Thanks for this work, but you have NOT reproduced the problem; possibly your fan does not evince the problem. Did you run my UNO test code?  This code:
`void loop() {
  Serial.println("Begin!");
 Serial.println(255);
  analogWrite(PWM_PIN,255);
  delay(D_MS); 
   Serial.println(220);
  analogWrite(PWM_PIN,220);
  delay(D_MS);
    Serial.println(200);
  analogWrite(PWM_PIN,200);
  delay(D_MS);
  Serial.println(190);
    analogWrite(PWM_PIN,190);
  delay(D_MS);
    Serial.println(80);
    analogWrite(PWM_PIN,80);
  delay(D_MS);
      Serial.println(40);
    analogWrite(PWM_PIN,40);
  delay(D_MS);
        Serial.println(12);
    analogWrite(PWM_PIN,12);
  delay(D_MS);
}`

...on an UNO, with the XY-GMOS board, but I doubt it is different, does NOT turn the fan at a pwm duty cycle or 200/255, 190/255, 80/255, etc.  However, those ratios of a steady 24V signal (down to about 8V) turn the fan smoothly and start it from a stop.  If you're fan will start at 50% duty cycle via PWM, then either my fan is different (which I believe), or I am terribly confused (which is possible but unlikely.)  In this case, I could either test it again or your can order my fan and try it for yourself.

ForrestErickson, 2023-05-31:
Regarding, "Did you run my UNO test code? This code:"  No just now found it. I was using the Arduino UNO code I have uploaded here to my Github:

https://github.com/ForrestErickson/FrequencyCounterusingTimers1and2/tree/main

ForrestErickson, 2023-05-31:
Notes regarding ordering fans an email from Lee to Robert

> Per out phone conversation 12:43 today I think we should evaluate side by side BOTH the fan with OUT PWM and the fan with PWM.
> Unfortunately neither Mouser nor Digikey has both in stock. But they each have the one the other does not.
>  
> Regarding, "9MB24P2K01"  For the fan with out WPM, I believe you meant to type "9BMB24K201"    Please confirm.
> 
> The equivalent fan with PWM input looks to be "9BMB24P2K01"  which is almost what you typed but with a "B" just after the first digit. Please confirm.
>  
> Digikey has the one with OUT PWM input in stock:
>  
> 1688-2166-ND
> 9BMB24K201
> Sanyo Denki America Inc.
> FAN BLOWER 97.1X33MM 24VDC WIRE
> 
> And Mouser has one WITH PWM in stock:
> 
> Mouser #:978-9BMB24P2K01
> Mfr. #:9BMB24P2K01
> Mfr.:Sanyo Denki
> 
> Regarding the Due,  I am sure I have a learning curve before I can use one but if I have one then where you can guide me I can better duplicate your setup so add to the Mouser order one please.  BTW, With what are these powered?
>  
> Mouser #:782-A000062
> Mfr. #:
> A000062
> Mfr.:Arduino
>  
> Robert, please order one fan each from each distributor.
> Ship to:
> Public Invention
> Care of Forrest Erickson
>  
>  
> Regarding the "The motor controller board ... XY-GMOS 10A MOSFET Module 1"
> The "Estimated delivery on Aug 01 " so not timly. I need one from Robert's stock so please SHIP TWO. (So I can blow one up and not be dead in the water.)
> 
> Robert,
> 
> Do you have any of these LM2596 DC-DC Buck Converter Step Down Module Power Supply Output 1.23V-30V or similar?
> Example: https://www.amazon.com/LM2596-Converter-Module-Supply-1-23V-30V/dp/B008BHBEE0/ref=asc_df_B008BHBEE0
> 
> I have some and may try to use them as a voltage follower however they will have a minimum drop of several volts so they will not power a fan to 24V off of a 24V supply.
>

ForrestErickson, 2023-06-05:
## Duplicating Robert's Three Wire Motor Difficulty
20230605 Lee received 9BMB24K201_ThreeWire motor.

Wired it up and tested with FanControlLinearized with a customized version V 0.7 which PWM to a motor control board.
This test is with the HiLetgo motor controller which is an L298N.

I found that the motor only spins with duty cycle of something like 220 out of 255.  Does not spin at 210 out of 255.

### Graphical Results
Test condition. PWM Sweep from 0 to 255.
For most of the PWM sweep the motor is not turning.
The FG (Tachometer) is reading the chopped power and so the count is way high.
Once the motor starts turning the count goes even higher then lower.
[image]

### Oscilloscope of some waveforms
Duty cycle well over 50% and yet motor not turning.
[image]

Duty cycle near 100% and motor turning but notice chopped FG signal
[image]

ForrestErickson, 2023-06-05:
Testing with XY-GMOS motor control module.

Was having hard time to get the control module to modulate the power into the FAN.  With out a schematic for using this module or for the internal circuit of this module I was perplexed.  I had connected the Vin= to the OUT- and the fan was on continiouslly.
It appears the OUT+ is tied to the VIN+ and it is the OUT- that is switched.  
I finally moved the Fan Ground wire to the OUT- and now the PWM can control the fan.
However, I get no FG (Tachometer) signal reading in the UNO.
The consequence of this is that the ground to the fan is switched and so the FG signal goes to +24V

I may have failed the UNO. I see the signal on CH2 is going up to 24V. WAY over the Vcc of 5V for the UNO.

Here is the PWM step from 0 to 255 in blue, and the +24 supply in Green. There is no FG (Tachometer)
[image]

LokiMetaSmith, 2023-07-12:
see #81 for more detail, but a PWM DC motor driver is not suitable for driving a BLDC fan

closed 2023-07-12 by LokiMetaSmith