PubInv/NASA-COG #81: Check that the 4-wire fan can be treated like the 3-wire fan by lowering voltage...

RobertLRead, 2023-06-02:
Dear Lee,

I don't have the PWM version of the fan, so I can't check this, but I suspect that most PWM fans will work fine by slowing down as the voltage is gradually decreased from 24V.

In that sense, a PWM fan can work just like the "3-wire" fan. This seems like it would be a painless and palatable solution for American Oxygen, because they could use a PWM fan and simply treat it as a 3-wire fan. 

Can you check that? I will make a Github issue of this.

ForrestErickson, 2023-06-02:
20230602 Lee has received 4 wire fan, 9BMB24P2K01
Was suppressed to find no connector on the end of the wires. Here is the wire identification as found in the specification:
[image]

Will have to find some kind of 0.100 header to solder on to the wires if I am to have pin compatible connections into my test apparatus which is currently set up for the four wire PC like fans.

RobertLRead, 2023-06-03:
I use alligator clips.

ForrestErickson, 2023-06-03:
Wiring up using some 2mm JST connectors I have. Then a JST to 0.100 adapter cable.
[image]

ForrestErickson, 2023-06-05:
## Next test by varying power supply.

I set for maximum PWM and full normal bench supply voltage (24V).
I set the UNO to measure the voltage and scaled it so that 2400 is 24V supply on this graph.  Supply Voltage is in green.
I then manually reduced the supply voltage and the RPM tracks but not quit linearly.
 
###  RPM Becomes Inaccurate
As you can see FG (Tachometer) signal (in red) becomes unreliable at about 8 volts.

[image]

### FG Signal When RPM Inaccurate
Here is oscilloscope waveform, Channel 2, of the FG signal. The amplitude has become low enough that it does not reliably trigger the Arduino UNO input pin when the fan supply voltage is 7.5Volts (channel 4)
[image]

### Accurate RPM at 8Volt
Increasing the fan supply to 8 volts makes the RPM accurate. 
[image]

### RPM signal at 24 Volts
[image]

ForrestErickson, 2023-06-05:
## Testing of 9BMB24K201_ThreeWire fan.
Repeat test by reducing the 24V DC supply gradually.
RPM and V Supply   
[image]

### Unstable / Inaccurate FG  
The FG signal became inaccurate was not stable once the voltage got below about 7.9V.
[image]
Green trace is 24V supply. Where 24V is 2400 on this scale.
Red trace is FG (Tachometer) in RPM.

ForrestErickson, 2023-06-05:
## Three Wire Fan Same As Four Wire Fan?  NO.

I have tried WITH OUT SUCCESS by adding a forth wire to the three wire fan, to PWM it.
Apparently there is more to it than the wire.

Photo showing the open pad to which I attached a wire and the PWM signal but with out success.
[image]

LokiMetaSmith, 2023-07-12:
I'm going to close this issue. We've determined that the "three wire" system with tach output is in fact a BLDC fan controller that is designed to operate at a fixed DC voltage. Modulating the DC voltage to enable speed control, interferes with the complex operation of the BLDC controller.

closed 2023-07-12 by LokiMetaSmith