PubInv/NASA-COG #84: Labeling, Identifying MAC on the OEDICOS

ForrestErickson, 2023-06-06:
While reading the An Oxygen Engine Digital Control System (DRAFT v.0.1) I encountered:

the most elegant way to do address and control hundreds or even thousands of Oxygen Engines is to create an Ethernet-based Local Area Network of OEDICOSs where each has its own unique local IP (Internet Protocol) address

This lead me to remember that for an IP device we need labeling for MAC addresses.

Related, is how does the installer and user(s) find the IP address for a particular OEDICOS ?

LokiMetaSmith, 2023-07-12:
https://docs.arduino.cc/tutorials/ethernet-shield-rev2/dns-web-client

MAC address can be defined in the arduino code, or the default one on the device can be used. 
/ Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = {  0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02 };

ForrestErickson, 2023-09-03:
I assume that at this time OEDCS Serial number 1 has the ethernet shield in place and that the MAC could be found and a label added to the device.

I assume that soon if not already the same for OEDCS Serial number 2.

Some documentation on how the MAC is read out and capture of the MAC some where our future selves can find it is prudent.

ForrestErickson, 2023-09-18:
Note that as of 20230918 the fiftieth line of serial output of the DUT includes the MAC address. For example

> 13:29:04.242 -> Starting Ox...
> 13:29:04.338 -> Create watchdog (todo)
> 13:29:04.338 -> Core booted...
> 13:29:04.478 -> BEGINNING ASSERTION CHECKS!!
> 13:29:04.478 -> IF YOU DO NOT SEE THE WORDS 'ALL CLEAR' BELOW AN ASSERTION HAS FAILED
> 13:29:04.573 -> ALL CLEAR!!
> 13:29:04.620 -> HAL: About to init Fan!
> 13:29:04.620 -> About to initialize heaters
> 13:29:04.714 -> HEATERS_INITIALIZED
> 13:29:05.844 -> didn't set address
> 13:29:06.927 -> didn't set address
> 13:29:06.927 -> Manuf: UNKWN
> 13:29:08.480 -> didn't set address
> 13:29:08.480 -> Model: 
> 13:29:10.015 -> didn't set address
> 13:29:10.015 -> VoltageSt: 
> 13:29:11.561 -> didn't set address
> 13:29:11.561 -> Rev: 
> 13:29:13.110 -> didn't set address
> 13:29:13.110 -> ManufDate: 
> 13:29:14.663 -> didn't set address
> 13:29:14.663 -> Serial: 
> 13:29:16.210 -> didn't set address
> 13:29:16.210 -> Country: 
> 13:29:17.739 -> didn't set address
> 13:29:17.739 -> RateVoltage: 65535
> 13:29:19.331 -> didn't set address
> 13:29:19.331 -> RateCurrent: 65535
> 13:29:19.798 -> MaxVoltage: 65535
> 13:29:20.314 -> MaxCurrent: 65535
> 13:29:21.862 -> didn't set address
> 13:29:21.862 -> failed to turn it on
> 13:29:22.894 -> didn't set address
> 13:29:22.894 -> failed to set volts
> 13:29:23.970 -> didn't set address
> 13:29:23.970 -> failed to set current
> 13:29:23.970 -> HAL:About to return!
> 13:29:23.970 -> Successful init of Hardware Abastraction Layer!
> 13:29:23.970 -> ReadTempsTask init
> 13:29:24.017 -> Locating devices...Found 2 devices.
> 13:29:24.017 -> Parasite power is: OFF
> 13:29:24.110 -> Unable to find address for Device 2
> 13:29:24.110 -> Device 0 Address: 3B38B7940C4C6CB4
> 13:29:24.110 -> Device 1 Address: 3B5AB9940C4C0CDE
> 13:29:24.110 -> Device 2 Address: 3B5AB9940C4C0CDE
> 13:29:24.204 -> Device 0 Resolution: 12
> 13:29:24.251 -> Device 1 Resolution: 12
> 13:29:24.251 -> Device 2 Resolution: 12
> 13:29:24.251 -> Config of temperature sensors done
> 13:29:24.251 -> CogTask init
> 13:29:24.251 -> TempRefreshTask init
> 13:29:24.817 -> MAC Address: FE:ED:16:6B:72:67

ForrestErickson, 2023-09-18:
I have labeled the MOM unit with the MAC address I have.

closed 2023-09-18 by ForrestErickson