PubInv/NASA-COG #87: Life Time of SanAce 9BMB Fan, 4.5 Years

ForrestErickson, 2023-06-23:
The SanAce 9BMB models of fans have a rated life of "40000/60˚C"  This is about 4.5 years.

This moving part and the fan in the ODECOS programable power supply represent moving parts which will have a predicted lifetime. 

The design of the COG system should provide for easy access to these fans.

See "Expected life" in the datasheet at:
https://www.mouser.com/datasheet/2/471/San_Ace_97BMB33_E-1284357.pdf

LokiMetaSmith, 2023-07-12:
that lifetime can accelerate or decelerate based on operating conditions. for example, if the unit is run at 30degC, then it would by spec operate for 9 years, which would be adequate.

ForrestErickson, 2023-09-03:
@RobertLRead 
Just thinking out loud. 
Delivery of the OEDCS to American Oxygen is a good time to report this expected life time parameter.

CC: @LokiMetaSmith

LokiMetaSmith, 2023-10-06:
In the manual we should list out replicable parts or parts that are considered consumables

closed 2023-10-06 by LokiMetaSmith

ForrestErickson, 2023-10-06:
Regarding, "In the manual we should "
Where in the repository is this manual ?

LokiMetaSmith, 2023-10-06:
I believe it's currently captured in the google doc "User Manual 0.3.5 and
The 5-Knob OEDCS Approach"

At some point it should be revisioned and converted to Markdown for
inclusion in the repository.