PubInv/NASA-COG #91: DIN Rail Location Change Request

ForrestErickson, 2023-07-03:
Robert's email of 7/2/2023 1:33 PM

> Dear Lee,
> 
> Here is a photo of the top, with some measurements of the current design. 
> 
> However, the current design is imperfect.  The DIN rail closest to the front is nearly impossible to use if the Anderson connectors are in place. I believe that that rail should be moved further away from the front by about an inch.
> 
> Likewise, the second DIN rail parallel to the front panel (the one further from the front panel) is too close to the power supplies for anything to be placed there when the power supplies are actually there (which, under NASA current but-ever-changing plans are not fully populated.)
> 
> I suggest that this rail be moved 2" closer to the front panel (I know this will put the two rails very close together and preclude a component in the same position horizontally on both rails.
> 
> Then I believe we should place another set of railes much further back, about 12" from he front panel. This will of course be unusable if the power supplies are all installed; but we currently imagine no more than 4 being installed.

[image]

ForrestErickson, 2023-09-03:
@LokiMetaSmith 

Please document the DIN rails of Serial Number 2 as it matures.

CC: @RobertLRead

LokiMetaSmith, 2023-10-06:
I don't think this is useful, but this is how the top low voltage din rail was used in OEDCS#2

[image]

closed 2023-10-06 by LokiMetaSmith