PubInv/NASA-COG #92: USB Connection or Changing Baud Connection Rate resets the DUE?

ForrestErickson, 2023-07-05:
Does the Due reset when the USB is connected or other conditions such as changing BAUD rate?

What will be the impact on the controlled "PLANT" in any and all operating conditions when the Due goes through a reset?

What will be the default condition of EVERY control output?  Is this optimum?

How long from reset before the control loop stabilizes?

It was the bug report on the GPAD (Changing baud connection rate resets the GPAD #242) which made me think of this.

ForrestErickson, 2023-07-05:
I chose the label this as dangerous until proven otherwise.

LokiMetaSmith, 2023-07-12:
This is an important distinction, we may need an analog/hardware based "safe shutdown" circuit that uses a "watch dog" keep alive from the processor in the event of a system critical freeze, a reboot, or reprogramming of the device. 

This makes the assumption that "shutting down" is always a safe option.

ForrestErickson, 2023-09-03:
@LokiMetaSmith 
@RobertLRead 

I think the status of this is that we know reset on the Due when the USB connection is established such as during firmware upload is expected normal Due behavior.

Although we know of a rework to change that behavior we have not decided to do that rework ( nor even evaluated it) Rather the firmware has been enhanced so that at reset the system state is evaluated and the system place under control in a safe manor.

IS THIS STATEMENT CORRECT?

Can we close this issue?

closed 2023-10-09 by RobertLRead