PubInv/NASA-COG #95: The current design circuitry provides power to the fan, and it runs full speed open...

RobertLRead, 2023-07-22:
the PWM fan runs full speed if the PWM wire is an open circuit. This occurs when the microcontroller starts up and the pin is initialized. Although brief, we clearly should not have the fan running when it is uncontrolled. The solution would appear to be to supply the power entirely with a motor controller or resistor. Possible a pull-down resistor will work, but this needs to be tested.

LokiMetaSmith, 2023-07-24:
Those won't work. The fan will need to be integrated in the soft/safe shutdown circuit. The always safe option is to have the fan OFF, which can be achieved with a power p channel mosfet that will need to be enabled before 24v can be applied to the fan.

ForrestErickson, 2023-08-21:
@LokiMetaSmith 
@RobertLRead 

In the document, "https://docs.google.com/document/d/1n3v-AxqNKEzcKqmHicGKnFmG8sKbNEL_pGaUlU9CZ8U/edit" 

The DB25 MALE connector has pins 2-5 as +12V.
I have to assume now that this is +24V. Further with pins 2-5 connected together the power is switched off together. This means that any and all fans are off at the same time and when +24V is applied the fans are turning no slower than the minimum.

Therefor in the doucment at 
https://github.com/PubInv/NASA-COG/tree/develop/elec/README.md

I am changing the table from
[image]

To
[image]

ForrestErickson, 2023-08-21:
Candidate P enhancement MOSFET: 
https://jlcpcb.com/partdetail/117243-NCE01P30K/C115991

Added to schematic on Fan power switch circuit.

ForrestErickson, 2023-08-21:
@RobertLRead 
I have identified the signal that will switch on and off the +12V Switched (fans) as: **FAN_POWER**
**Please identify the Due pin to which you want this control routed.**

Note the naming convention.  Asserted high turns on power to the fans.
This net when high impedance (Due in reset) will be pulled low and fans will be off. 

After a reset / power up the control system should determine the most appropriate value for this pin per what ever states must be evaluated.

ForrestErickson, 2023-08-21:
@RobertLRead ,
Once you have identified a Due pin for control of FAN_POWER this issue can be closed but not before.

ForrestErickson, 2023-08-21:
I have pushed changes to the PCB which should address this and Test and understand the Tachometer Feature of the 24V blower with both code and an oscilloscope #75

ForrestErickson, 2023-08-31:
@LokiMetaSmith 
@RobertLRead 

Done but this should be tested when Controller V1 assemblies are integrated into Serial Number 1 and Serial Number 2.

I will close this task because the circuit design has been done. 
A new task for Robert will be confirming, testing the functionality of Controller V1 printed wiring assembly.

ForrestErickson, 2023-09-03:
I will close this task because the circuit design has been done.
A new task for Robert will be confirming, testing the functionality of Controller V1 printed wiring assembly.

closed 2023-09-03 by ForrestErickson