PubInv/NASA-COG #103: TF800A12K unavailable

LokiMetaSmith, 2023-07-31:
https://octopart.com/search?q=TF800A12K&currency=USD&specs=0

Digikey and Mouser are out of stock on this item. 

Will need to identify a substitute that works with existing firmware or find an alternative that is more reliable to source.

LokiMetaSmith, 2023-07-31:
Price has also balooned from $377.26 to Unit Price:$537.390 /Each

ForrestErickson, 2023-07-31:
TF800A12K

https://slpower.com/contact
California USA
SL Power Headquarters
27001 Agoura Rd. Suite 325
Calabasas CA 91301
United States
[+1 (888) 412-7832](tel:1888-412-7832)

Massachusetts USA
Sales & Engineering Office
6 Merchant St. Suite 2
Sharon, MA 02067
United States
[+1 (888) 412-7832](tel:1888-412-7832)

Shanghai China
Sales & Engineering Office
4th Fl. Building 53 1089
Qing Zhou Road North
Shanghai, China 200233
[+86 29 8874 1895](tel:+86-29-8874-1895)

Hatfield UK
Sales & Engineering Office
1 Travellers Ln.
Welham Green Hatfield
Hertfordshire AL9 7JB

For TN distubitor
Digikey
Mouser
SAGER 407-833-8833
WPG America 978-657-0024
And in TX the same and in addition:
Avnet 281-243-7330 
Arrow  832-327-4216

ForrestErickson, 2023-07-31:
Lee called AGER 407-833-8833. Caller ID says Lake Mary FL
There isn't any stock or on order.  22 piece min buy.
None in stock anywhere.  

They have a power supply department that does custom supplies
Power systems engineer on staff in Texas.   Call: 469-467-1305 or 508-947-8888  Might be Don Baldwin ....  dbaldwin@sager.com

Lee called and left voice mail at: 508-947-8888 to Don Baldwin

ForrestErickson, 2023-07-31:
Lee called WPG America 978-657-0024. 
Left voice mail.

ForrestErickson, 2023-07-31:
Lee called Avnet 281-243-7330.
Press 0 and left voice mail.

ForrestErickson, 2023-07-31:
Other ideas

> [Newpowersupply@outlook.com](https://connect.xfinity.com/appsuite/#)
>  Hello New Power Supply,
>  
> I saw your product at:
> https://www.aliexpress.us/item/3256802295343621.html?gatewayAdapt=glo2usa4itemAdapt#nav-review
>  
> We are interested in learning more for our application of 12V at up to 60A
> We need programable voltage  and current limit so we want the feature:
> Customize#6 RS232/RS485 Communication port control by computer
> We may also want the 19" rack mount customization.
>  
> Can you send us a user manual / programmers manual we can examine to understand more about this power supply so we can decide if it meets our needs?
>  
> After we find a supply which meets our needs, we hope to find and purchase one or two supplies immediately and then perhaps 10 more in a few months for a custom industrial control application.
> 
> We will want UL or similar safety certification.
>  
>  
> Thanks you,
> (Forrest) Lee Erickson
> Product designer, Public Invention
>  
> CC Lawrence and Robert
>

ForrestErickson, 2023-08-01:
@RobertLRead 
@LokiMetaSmith 
I am collecting the traffice regarding th SL Power Electronics, TF800A12K into the issue.

> On 07/31/2023 6:52 PM EDT Robert Read <read.robert@gmail.com> wrote:
> This link came from the Octoparts link that Lawrence sent me:
> https://www.verical.com/pd/sl-power-ac-to-dc-power-supply-TF800A12K-7272394?
>tm_campaign=octopart_2022&utm_content=inv_listing&utm_currency=USD&utm_keyword=TF800A12K&utm_medium=aggregator&utm_source=octopart

> On 07/31/2023 7:34 PM EDT FORREST ERICKSON <forresterickson@comcast.net> wrote:
> The manufacturier of the supply told me today that Arrow is a distributor and this group seams to be part of Arrow.  See the US phone number below.
> Who calls them tomorrow?  You, me, Lawrence, !!! So many options.
> 

> On 07/31/2023 10:53 PM EDT Lawrence Kincheloe <lokimail@gmail.com> wrote:
> Vertical is a supply partner marketer. I've not used them, so I can't comment on them. 
> Essentially, they put unused inventory from "supply partners" who are ISO9001 certified companies. 
> So essentially you're buying unused inventory that a company has on the books but isn't equipped to sell. 
> It's probably fine and the price is right, but supply is NO way guaranteed, and if AmOx turns around and says "I want 10 more units" that won't be possible. 
>

ForrestErickson, 2023-08-01:
@RobertLRead
@LokiMetaSmith

 I believe we should purchase their entire stock.
 I believe we then need to follow up with the manufacturer with an order for a few more to try to encourage them to refill the pipeline.  We tell them point blank we have purchased all of the apparent stock and anticipate and hope for a few more requirements.  

With the one we have and these ten we will still be short about one or three but we will have time for them to build.

ForrestErickson, 2023-08-01:
Vertical is probably operating on Mountain Time
Company Address
Arrow Electronics, Inc
9201 East Dry Creek Road
Centennial, CO 80112

ForrestErickson, 2023-08-01:
Regarding the email I sent to "Newpowersupply@outlook.com Hello New Power Supply,"

> 发件人: FORREST ERICKSON <forresterickson@comcast.net>
> 发送时间: Tuesday, August 1, 2023 5:11:55 AM
> 收件人: Newpowersupply@outlook.com <Newpowersupply@outlook.com>
> 抄送: Lawrence Kincheloe <lokimail@gmail.com>; Robert Read <read.robert@gmail.com>
> 主题: Regarding, Model NP-ATDYN1260 12V 60A 720W Adjustable DC Laboratory Power Supply
>  
> Hello New Power Supply,
>  
> I saw your product at:
> https://www.aliexpress.us/item/3256802295343621.html?gatewayAdapt=glo2usa4itemAdapt#nav-review
>  
> We are interested in learning more for our application of 12V at up to 60A
> We need programable voltage  and current limit so we want the feature:
> Customize#6 RS232/RS485 Communication port control by computer
> We may also want the 19" rack mount customization.
>  
> Can you send us a user manual / programmers manual we can examine to understand more about this power supply so we can decide if it meets our needs?
>  
> After we find a supply which meets our needs, we hope to find and purchase one or two supplies immediately and then perhaps 10 more in a few months for a custom industrial control application.
> 
> We will want UL or similar safety certification.
>  
>  
> Thanks you,
> (Forrest) Lee Erickson
> Product designer, Public Invention
>  
> CC Lawrence and Robert
 

I got the following reply / quote:

> On 08/01/2023 7:50 AM EDT yu Abby <newpowersupply@outlook.com> wrote:
> Hello
> (Forrest) Lee Erickson
> 12VDC 60A
> Rack mounted cost:
> 1、Unit price: 506 US dollars/unit
> 2、Customized RS232/RS485 cost: $85 per unit
> 
> You can refer to the following links for rack mounted enclosures and testing videos.
> 
> https://www.aliexpress.us/item/3256802295343621.html?gatewayAdapt=glo2usa4itemAdapt#nav-review
> Ms. Abby Yu  
> Cell & Whatsapp: +86-15914089745
> Tel: +86-20-89282095
> QQ: 2524550484
> E-mail: Newpowersupply@outlook.com
> 
>

LokiMetaSmith, 2023-08-01:
Possible other replacements that are open source

- https://www.envox.eu/bench-power-supply/introduction/    **Looks to use a linear transistor to provide voltage regulation, won't do high current, but fully open**

LokiMetaSmith, 2023-08-01:
Possible replacements,

https://www.digikey.com/en/products/detail/xp-power/HDS800PS12/5023629

ForrestErickson, 2023-08-01:
Suggestion Robert purchase 1 of the SL Power Electronics from Verical .
Robert purchase 1 of the XP supply from digikey **to test form and function compatibility with SL Power Electronics.**
Lawrence to purchase SL Power Electronics from Verical.

RobertLRead, 2023-08-01:
I have ordered an 1 XP power supply and 1 SL power supply

LokiMetaSmith, 2023-08-02:
I've also ordered 1xp and 1sl power supply as well as the following.

[image: image.png]

ForrestErickson, 2023-08-07:
@LokiMetaSmith, 
@RobertLRead ,

Regarding, " Depending on the data communication verification,"
I am sure some where I have down loaded the documentation for programing the ST Electronic power supply.
I do not know if it has been commited to the repoistory some where but we should have the full document in perhaps a reference folder or a link to such documentation.
To this we idealy add the similar document or a link for the XT power supply.

If documents, I think a refrence folder logicaly belongs in the https://github.com/PubInv/NASA-COG/tree/develop/firmware folder.

If only a link then in a README.md file there.

ForrestErickson, 2023-08-07:
Here is a link to the ST Power Electronics programing PDF
https://slpower.com/data/collateral/AN-P0036%20TF%20Series%20Communication%20Protocal%20User%20Manual.pdf

LokiMetaSmith, 2023-08-07:
something non-obvious that i've observed in other high voltage power
supplies, sometimes the digital control pins aren't referenced to earth
ground and computers often are. This can create a ground reference voltage
drop somewhere around 70+ volts. I've destroyed at least one programmer
this way, and it's something to be aware of when considering powering the
units.

ForrestErickson, 2023-08-07:
Email from 
On 08/06/2023 12:17 PM EDT Lawrence Kincheloe <lokimail@gmail.com> wrote:
Units are nearly identical but come from different manufacturers. I suspect that the design was licensed. This means that physically they will be compatible, but firmware may differ. Will investigate further when next digikey order arrives

ForrestErickson, 2023-08-07:
@LokiMetaSmith ,

Regarding, "something non-obvious that i've observed in other high voltage power
supplies, sometimes the digital control pins aren't referenced to earth
ground and computers often are. ..."

You should resolve this with a multi meter as soon as possible. Does this need it's own Issue or can you jump on it right away?
And if grounds are seperate our wiring must connect them in the best possible way (TBD).

ForrestErickson, 2023-09-03:
Closing as we were able to order both the SL Power Electronics part and the XP Power supply part and they so far have been found to be interchangeable.

closed 2023-09-03 by ForrestErickson