PubInv/NASA-COG #114: Standardize Interface to Plant

ForrestErickson, 2023-08-09:
The connections from the ODECOS to the plant are defined at the ODECOS at the ODECOS front panel.

For development of the plants that might be controlled for test and development (dummy plants) would be more interchangeable if we had a "plant front panel" or "PanelPlantFront" or similar

ForrestErickson, 2023-08-09:
Lee requested from Omega drawings for MSS-04
https://www.omega.com/en-us/contact-us

> Need full mechanical dimensions for MSS-04.  I am designing the panel for a device which will use this and need the exact hole locations which are not provided on your drawing "MSS_Series.pdf"
> 
> I would like 3D models of the MSS-04 as STEP files (FreeCAD would be even better) or STL if that is the best you can do.
>

ForrestErickson, 2023-08-09:
IEC power receptacle might be handy.

<html>
<body>

Digi-Key Part Number | 486-1003-ND |  
-- | -- | --
Manufacturer | SCHURTER Inc. |  
Manufacturer Product Number | 6100.3300 |  
Description | PWR ENT RCPT IEC320-C14 PANEL QC |  
Detailed Description | Power Entry Connector Receptacle, Male Blades IEC 320-C14 Panel Mount, Flange |  

<br class="Apple-interchange-newline">
</body>
</html>
Digi-Key Part Number
486-1003-ND
Manufacturer
[SCHURTER Inc.](https://www.digikey.com/en/supplier-centers/schurter)
Manufacturer Product Number
6100.3300
Description
PWR ENT RCPT IEC320-C14 PANEL QC
Detailed Description
Power Entry Connector Receptacle, Male Blades IEC 320-C14 Panel Mount, Flange

ForrestErickson, 2023-08-09:
@RobertLRead wrote

Perhaps an even better idea would be to treat that constellation of components as a module. In our current enclosure, we could probably pack 4 in.

If the constellation were made long and skinny, it might fit into a 1U blade.

ForrestErickson, 2023-08-09:
Found...
While server sizes may differ slightly depending on the make and model, a 1U server typically occupies a dimension of 1.75″ by 19″ of rack space. There are also 2U and 4U servers, which offer more computing power but take up more space on the rack.

ForrestErickson, 2023-08-09:
Powerwerx 

Hi Forrest,

> 
> Unfortunately we do not have these files for those items.
> 
> Sincerely,
> Team Powerwerx

>

ForrestErickson, 2023-08-09:
Useful dimension information here:

https://en.m.wikipedia.org/wiki/Rack_unit

LokiMetaSmith, 2023-08-10:
@ForrestErickson consider using C-19 and C-20 connectors rated for 20amps at 240vAC

https://www.interpower.com/ic/InfoPower/iec-60320-c19-c20.html

we can panel mount a C20 inlet on the CFC oxygen plant panel, and a shrouded C19 on the ODECOS.

Cables are an off the shelf item

ForrestErickson, 2023-08-10:
@LokiMetaSmith Regarding C-19 and C20. Nice find. 
FYI I had put two of the 10A connectors in parallel as an idea for using what I knew.
[image]

ForrestErickson, 2023-08-10:
Here is a mechanically similar panel mount C20 in stock at Digi-key
https://www.digikey.com/en/products/detail/schurter-inc/6163-0004/2646990

[image]