PubInv/NASA-COG #117: wrong cutout or missing part informatin

LokiMetaSmith, 2023-08-10:
12 position thermocouple window should be  20.1 x 167.6 (0.79 x 6.60)  mm(inch)
for MSS-12 

panel cutout is 27.45 x 174 mm 

Missing the panel assembly plate part number to go along with the MSS-12 and [MPJ](https://www.omega.com/en-us/temperature-measurement/temperature-connectors-panels-and-block-assemblies/panel-connectors-and-assemblies/p/MPJ)

LokiMetaSmith, 2023-08-10:
[image]

ForrestErickson, 2023-08-10:
Perhaps I am missing something.
The cut out on the Frontpanel provides and is intended to provide clearance. The MSS-12 part has the tight tollerence dimensions for retention of the thermocouple mini jack connectors.  
Have you ordered the MSS-12 part? 

Back in March we did not intend to cut the panel to receive the mini jacks directly. Unit 2 should be built the same way.

RobertLRead, 2023-08-10:
I think the jack pane can be ordered here:

[https://assets.omega.com/pdf/connectors/jack-panel-assemblies/panel-jack-assemblies/MJP.pdf](https://assets.omega.com/pdf/connectors/jack-panel-assemblies/panel-jack-assemblies/MJP.pdf)

I don't believe I got good dimensions, I measured them myself and told them to Lee.

LokiMetaSmith, 2023-08-10:
Omega is out of stock, but
MJP1-12-K is available from Newark

LokiMetaSmith, 2023-08-10:
Part ordered, hopefully it's fully assembled and doesn't need additional
hardware.

ForrestErickson, 2023-09-03:
@LokiMetaSmith 
IIRC you have indicated that there is a good fit on Serial number 2 as fabricated.
Please document and close out this issue as appropriate.

LokiMetaSmith, 2023-10-06:
can confirm the fit was good on this part.

closed 2023-10-06 by LokiMetaSmith