PubInv/NASA-COG #118: Replace panel GFCI with system wide ground fault leakage current detector breaker

LokiMetaSmith, 2023-08-10:
can use a single gfci to cover the entire enclosure.

https://en.wikipedia.org/wiki/Residual-current_device#:~:text=Residual%2Dcurrent%20and%20over%2Dcurrent,protection)%20in%20Europe%20and%20Australia.

LokiMetaSmith, 2023-08-10:
This would be a Main breaker for the entire system, this would be rated for a 40 amp circuit
Cost is a bit high at, $214.94

https://www.newark.com/schneider-electric/a9r41240/earth-leakage-protector-2p-40a/dp/66AH5315?st=earth%20leakage

We may need to discuss the cost/benifit of GFCI in the future

RobertLRead, 2023-08-10:
You may have to educate me on this. It seems to me that for the duration of
our contract the safety benefits are worth the price; but I admit to being
a bit hazy on the benefits of GGCI.

ForrestErickson, 2023-09-03:
@LokiMetaSmith ,
Please photograph as implemented in Serial number 2.
Please identify part numbers associated with this issues as implemented.
Please identify wiring diagram(s) as implemented.

LokiMetaSmith, 2023-10-06:
It was determined that different GFCI breakers have different performance when being installed (ODECS#2 used a Leviton brand GFCI), furthermore it was determined that it may not be necessary for future configurations of ODECS.

RobertLRead, 2023-10-06:
It's spelled OEDCS.

closed 2023-10-06 by LokiMetaSmith

LokiMetaSmith, 2023-10-08:
> @LokiMetaSmith <https://github.com/LokiMetaSmith>
> Regarding, "It was determined that different GFCI breakers have different
> performance when being installed (OEDCS#2 used a Leviton brand GFCI),..."
>
>    - What was the difference?
>    - How was it determined, by what measurement? How would I reproduce
>    the measurement?
>    - By what criteria was the difference determined relevant? Against
>    what measurement?

- What was the difference?
   Leviton has a built in function where the GFCI won't engage until power
   is fully engaged. So when the GFCI is first installed, it won't engage the
   heater until full power is applied to the GFIC. This is a safety feature.
   This presents a problem when we implement a ramp because we won't be able
   to fully turn on a SSR if the temperature profile of the ramp is working
   correctly.

   - How was it determined, by what measurement? How would I reproduce the
   measurement?
   When building up OEDCS#2 the lamp wouldn't turn on until the SSR reached
   a full ON state and the reset switch could be triggered
   - By what criteria was the difference determined relevant? Against what
   measurement?
   - The functionality of a GFCI makes sense in context of a consumer
   device, but for an industrial device, the value of having one installed
   isn't as clear.