PubInv/NASA-COG #120: Design control board interconnect pcb

LokiMetaSmith, 2023-08-11:
This board should incorporate features from the existing odecos configuration, and add features nessisary to connect to the plant panel

LokiMetaSmith, 2023-08-11:
Initial board schematic has been created and uploaded to the repo

ToDo
- [ ] Wire up the power supply labels to the Due
- [ ] Add hookups for thermocouples
- [ ] implement proper fan control circuit
- [ ] add SSR control lines and connector
- [ ] add mounting hardware points
- [ ] add DB9 and DB25 
- [ ]

ForrestErickson, 2023-08-24:
Lots of progress on 23 and 24th of August.
[image]

ForrestErickson, 2023-08-24:
@LokiMetaSmith 
@RobertLRead ,

Regarding, " Wire up the power supply labels to the Due"
Is this the +12 that is simply wired into the Due?  Am I to infer that the Due has an onboard regulator that can dissipate the heat that will result from a +12 input power?

I am left wondering, **What is the current on Unit1 at the Due input?**

ForrestErickson, 2023-08-31:
This task is done with the release and order of PWA from JLCPCB on about 20230828
[image]

ForrestErickson, 2023-09-03:
Closing since the PCB design was released and ordered on about 20230828.
If there are problems found with the Controller V1,  issues should be created at that time.

closed 2023-09-03 by ForrestErickson