PubInv/NASA-COG #121: Selection of SSR's for varying voltage range and their control

LokiMetaSmith, 2023-08-11:
In examining multiple options for SSR's

This one is idea, but is rated for 4v switching min, so would need a higher toggling voltage
[RSSDN-50A](https://www.newark.com/idec/rssdn-50a/ssr-din-panel-mount-660vac-32vdc/dp/13M2781)

alternatively, this one should work at 3.3v.
https://www.digikey.com/en/products/detail/carlo-gavazzi-inc/RM1A23D50/7696494

We should add the option of controlling SSR's with DC power supply instead of arduino 3.3v GPIO lines

LokiMetaSmith, 2023-08-11:
after writing up the requirements, it occurs to me that there is no need for a 50 amp SSR, however the flexibility of using different control voltages remain a feature request

ForrestErickson, 2023-08-14:
Interesting note in the specification:

> When the SSR output is subjected to a higher than rated voltage, a varistor or other element should be connected to the output terminals to absorb the over-voltage.

We should probably have MOVs some where in the system and I am years and years away from having read specifications for such.   There are also some arc gap snubber solutions too.

ForrestErickson, 2023-08-14:
@LokiMetaSmith 
Regarding, "it occurs to me that there is no need for a 50 amp SSR"
What is the exact part you have ordered?  or remind me again where I go look for that.
The exact heat sink too please.

LokiMetaSmith, 2023-08-14:
2
‎3667-SO943460-ND‎
‎SO943460‎
SSR RELAY SPST-NO 40A 12-280V
2
Immediate
51.51000 $103.02

3
‎3667-SO942440-ND‎
‎SO942440‎
SSR 25A/12-280VAC/TVS/CTRL 3-32V
1
Immediate
47.88000 $47.88
and heat sinks
https://www.amazon.com/dp/B08ZCYBPTV

ForrestErickson, 2023-08-18:
@LokiMetaSmith 

Just currious.
Regarding the heat sink:
[image]

Have you found the thermal resistance specified for it?
The SSR specifications 
[image]
show curves as a funtion of the heat sink thermal resistance.

ForrestErickson, 2023-09-03:
@LokiMetaSmith 
Please photograph the exact parts with which Serial number 2 is constructed. 
Identify the exact part numbers used please.

LokiMetaSmith, 2023-10-06:
[image]

closed 2023-10-06 by LokiMetaSmith