PubInv/NASA-COG #124: Physical Barriers Against Line Voltage

ForrestErickson, 2023-08-14:
Such as but not limited to the SSR.
Screw terminasl on programable power supply.
and other line voltages.
Fixed Due supply.

ForrestErickson, 2023-08-14:
The SSR specification refers to " Optional fingersafe terminal cover (RSS-CVR). 
Here is the Newark part:
https://www.newark.com/idec/rss-cvr/relay-accessory-cover/dp/31B3390?ost=rss-cvr
 "

RobertLRead, 2023-10-09:
@ForrestErickson I'm not sure where we are here. the OEDCS SN#1 does have a different finger-safe access cover installed. What should be done with this?

ForrestErickson, 2023-10-09:
@RobertLRead 
@LokiMetaSmith 
The scope of this was written with OEDCS Serial number 1 and 2 in mind.  Wanting to keep from electrocuting one another during development, but assuming knowledgeable users only would be poking around inside.

For a device to be sold, design for full safety should be the goal.  This means design for compliance to UL 950, (IIRC, for US domestic) and the corresponding European Norm for safety.