PubInv/NASA-COG #127: Missing part number for grounding bar

LokiMetaSmith, 2023-08-15:
The README.md for the front panel doesn't mention anything about the grounding bar, the hardware needed to attach it, or the part number/purchase location

LokiMetaSmith, 2023-08-16:
Grounding bar is compatible with Model #TGK12CP GE ground bar.
https://www.lowes.com/pd/GE-1-in-Ground-Bar-Kit/1001535

Screws provided are M5 and are compatible with a slotted or square driver

ForrestErickson, 2023-08-16:
Yes that is the part and I did purchase from Lowes.

I have updated the table on the README.md at: https://github.com/PubInv/NASA-COG/blob/develop/mech/frontpanel/README.md

I also added details for the thermocouple jack panel, snap strip and Jacks.

| 1        | ground bar                                   | GE                   | TGK12CP              | Lowes          | 105748                  | https://www.lowes.com/pd/GE-1-in-Ground-Bar-Kit/1001535                           | $8.98     |
| 1        | Jack Panel for Thermocouple 1x12             | Omega                | MJP1-12-BLANK        | Omega          | MJP1-12-BLANK           | [Jack Panel Drawing](Parts/Mech/mpj-j-f-thermocouple/JackPanel_MJP.pdf)           | $26.76    |
| 1        | Snap Strip for Thermocouple  1x12            | Omega                | MMS-12               | Omega          | MSS-12                  | [SnapStrip Drawing](Parts/Mech/MSS_Series.pdf)                                    | 12.34     |
| 3        | Miniature Thermocouple Panel Jack Connectors | Omega                | MPJ-K-F              | Omega          | MPJ-K-F                 | [SnapStrip Drawing](Parts/Mech/MSS_Series.pdf)                                    | $4.26     |

closed 2023-08-16 by ForrestErickson