PubInv/NASA-COG #130: Consider using American Oxygen's PID controller directory

RobertLRead, 2023-08-16:
American Oxygen seems to have written its own PID controller. This code runs on a RaspPI, but is in C++l; we could easily import it and use it in place of the PID_v1 standard arduino libary that we use. I don't think that matters very much, but it could be nice for AmOx because we would literally be using the same code (albeit on an Arduino), which would give them some familiarity (and consistency.)  Additionally, I will attempt to copy there proportional, integrative, and derivative constants (though Tom is using a 100.0 to 0.0 scale, and I use 1.0 to 0.0 to the duty cycle.)

https://bitbucket.org/americanoxygen/birch/src/master/src/amoxlib/src/pidControllerInterface.cpp

LokiMetaSmith, 2023-08-16:
we don't have access to that repo, can you fork the birch project, set it
as private and put it under PubInvention?

RobertLRead, 2023-08-16:
> Assigned #130 <https://github.com/PubInv/NASA-COG/issues/130> to

Tom gave me access to it.

LokiMetaSmith, 2023-08-16:
yes, but since I can't view it, I can't evaluate it for consideration when
writing the heater code

Please at your convenience, fork a copy, change the git server information
in .git/config and upload it to github in a private repo that Lee and I
have access to,

Or, download it as a zip, and share it directly and I can take it from
there.

RobertLRead, 2023-08-16:
I'd prefer not to do that; that might easily be misinterpreted by AmOx.

Let me just ask Tom to add both you and Lee to it.

It seems that Tom copied some existing PID controller, in any case.

RobertLRead, 2023-09-03:
Closing as irrelevant.

closed 2023-09-03 by RobertLRead