PubInv/NASA-COG #131: Holes for Grounding Bar Mounting, M5 or Clearance for M5?  Beyond SN 1,2?

ForrestErickson, 2023-08-16:
Regarding, 

> Grounding bar is compatible with Model #TGK12CP GE ground bar.
> https://www.lowes.com/pd/GE-1-in-Ground-Bar-Kit/1001535
> Screws provided are M5 and are compatible with a slotted or square driver

IIRC, The holes in SN1 were drilled and tapped for ???? screw
The holes in SN2 were specified for 5. 9mm so therefor a nut will be needed for mounting.

What should we do for future Front Panel ground bar?

LokiMetaSmith, 2023-08-16:
I used spare t-nut M5 nuts, a proper washer and nut would be ideal.

ForrestErickson, 2023-08-18:
@LokiMetaSmith 
What was the exact screw that came with the grounding bar?  I am doubtful it was metric.
Can you locate them?

ForrestErickson, 2023-09-03:
@LokiMetaSmith 
Please document this as you build up Serial number 2.
Some photos will be of help too.

LokiMetaSmith, 2023-10-06:
this will need to be optimized for future builds

closed 2023-10-06 by LokiMetaSmith

LokiMetaSmith, 2023-10-06:
They were indeed metric.

LokiMetaSmith, 2023-10-06:
> @LokiMetaSmith <https://github.com/LokiMetaSmith> ,
> Regarding, "this will need to be optimized for future builds"
> Please detail what was so wrong as to require a change.

Future builds should optimize the ground lug for just the number needed,
the DIN rails used include a ground bar so they have different
requirements, and we'll need to establish or find a method of grounding
shielded TC probes,

In short, we likely won't use the same ground bar.

ForrestErickson, 2023-10-06:
Is there a photograph covering the ground bar on Serial number 2?

Such photo could be added to this markdown.
https://github.com/PubInv/NASA-COG/blob/develop/mech/enclosure/Serial_number2/README.md

Same with +12 and +24 power supply location and wiring.

LokiMetaSmith, 2023-10-06:
Once the middle plate was installed, taking good photographs became
impossible.