PubInv/NASA-COG #135: evaluate panel mods for inclusion into freecad design files

LokiMetaSmith, 2023-08-16:
referencing #133   these changes weren't added to freecad drawings, but directly to the DXF files

This was a hack for expediency, the changes can be added back to the freecad drawings either after I've evaluated the fit and function of the end result or when time allows.

ForrestErickson, 2023-08-31:
Capturing some history from emails:

On Wed, Aug 16, 2023 at 4:43 PM FORREST ERICKSON <[forresterickson@comcast.net](mailto:forresterickson@comcast.net)> wrote:
Lawrence,
Re ah hack. I understand.
 
Please make an issue for updating the FreeCad drawing to your modified DXF. Be descriptive.
 
Regardless of the fit we should bring the FreeCad drawing into agreement with the parts ordered before we loose track of where we are.
 
Please identify the exact file with which the order(s) were placed.
 
Lee
On 08/16/2023 5:05 PM EDT Lawrence Kincheloe <[lokimail@gmail.com](mailto:lokimail@gmail.com)> wrote:
 
 
I've pushed the changes to the repo, note these changes aren't to the freecad drawings, but directly to the DXF files

This was a hack for expediency, the changes can be added back to the freecad drawings either after I've evaluated the fit and function of the end result or when time allows.

ForrestErickson, 2023-09-03:
It appears to @ForrestErickson  that the files affected are:

* [rear panel edit 1_3.dxf](https://github.com/PubInv/NASA-COG/blob/develop/mech/frontpanel/rear%20panel%20edit%201_3.dxf)
* [BottomPlateV1 edit.dxf](https://github.com/PubInv/NASA-COG/blob/develop/mech/frontpanel/BottomPlateV1%20edit.dxf)

and the associated SVG files.

LokiMetaSmith, 2023-10-06:
The added holes were useful, but the hole sizes shouldn't be used. Future builds should include panel mount considerations for Ethernet and USB interfaces.

closed 2023-10-06 by LokiMetaSmith