PubInv/NASA-COG #141: Make Markdown to capture "Due Pin-out map for OEDCS-1"

ForrestErickson, 2023-08-18:
From Robert's email:

I agree. Make an issue for me (or don't) and I will download the document into markdown.

On Fri, Aug 18, 2023 at 8:41 AM FORREST ERICKSON <[forresterickson@comcast.net](mailto:forresterickson@comcast.net)> wrote:
 
This documentation outside of the [https://github.com/PubInv/NASA-COG/tree/develop ](https://github.com/PubInv/NASA-COG/tree/develop)repository represents a failure to open source the information essential for a third party to benefit from the project.

The information of this document "Due Pin-out map for OEDCS-1" and other documents we are struggling to develop now should be captured in the project repository. 
 
A logical place might be this folder: https://github.com/PubInv/NASA-COG/tree/develop/elec
To make a start, I have made a README.md and place there a link to the google doc spread sheet.
 
All of the information in that google document could be in a stand alone markdown document.
 
Lee

RobertLRead, 2023-08-19:
This is done. I saw you put in some photos, but I did not include because I feel now that they are misleading, especially without the consideration of: [https://github.com/PubInv/NASA-COG/blob/develop/fancontroller/SanyoAceBreakout.fzz](https://github.com/PubInv/NASA-COG/blob/develop/fancontroller/SanyoAceBreakout.fzz)

ForrestErickson, 2023-08-19:
I have added more information to that page at https://github.com/PubInv/NASA-COG/tree/develop/elec

There is a table there which has a colum for the Due pins which is very incomplete.

ForrestErickson, 2023-09-03:
The table at the README.md link above is now very complete.
Closing since the PCB design was released and ordered on about 20230828.
If there are problems found with the table issues should be created at that time.

closed 2023-09-03 by ForrestErickson