PubInv/NASA-COG #143: Need Public Invention Wide KiCad Schematic and PCB Footprint Library

ForrestErickson, 2023-08-18:
The GPAD library for schematic symbols and PCB footprints represent a significate effort for parts management.  
These parts can be used with the confidence GPAD gave us that symbols and footprints are correct.
Additionally the creation of BOMs for ordering from JLCPCB with a spread sheet tools in the GPAD folder depends on the fields and the exact naming that this library has.

The Moonrat project recently got a copy of these libraries and is diverging.
The ODECOS Controller project is now facing the same problem. Divergence will happen.

ForrestErickson, 2023-08-18:
Lawrence suggested we have a separate repository for the master libraries of schematic symbols and PCB footprints.
This repository would be pulled into each project as a "subproject" which I think is a git terminology (There be dragons here)
Changes made by each user of the library can be local and then pushed back upstream.

ForrestErickson, 2023-08-18:
Link to the GPAD schematic symbol: 
https://github.com/PubInv/general-purpose-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/GPAD_SCH_LIB.kicad_sym

Link to the GPAD PCB footprints folder:
https://github.com/PubInv/general-purpose-alarm-device/tree/main/Hardware/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB.pretty

ForrestErickson, 2023-08-18:
Article about sub-repositories: https://dev.to/jjokah/submodules-a-git-repo-inside-a-git-repo-36l9

Lee has created a repository at: https://github.com/ForrestErickson/PublicInvention_KiCad_Library

This repository should be cloned into to PublicInvention as a stand alone project.
I think we need @RobertLRead  to do so.

ForrestErickson, 2023-08-18:
Now I am adding submodule to the NASA-COG/elec/Control Board/Control v1 folder in my clone of my fork.

[image]

[image]

ForrestErickson, 2023-08-18:
In KiCad lets use the symbol library manager to add schematic symbols.

[image]

[image]
Navigate to the new folder and select 
[image]
Select OK
[image]

ForrestErickson, 2023-08-18:
Now for Manage Footprint Libraries.

[image]

Select to Add Existing
[image]

Navigate too and select the folder
[image]

And there it is!
[image]
HOWEVER I did not check the make active box. 
And when I opened the footprint editor the new library is not there. SAD FACE HERE.

ForrestErickson, 2023-08-18:
Opened the footprint manager and selected the make active box and save and got this error.

[image]

HOWEVER the footprint library shows up in the editor!

[image]

ForrestErickson, 2023-08-18:
These libraries are working.

@RobertLRead 
So that Moonrat and the OEDCS (and retroactively GPAD) can share common KiCad libraries, please make a PubInv repository "[PublicInvention_KiCad_Library](https://github.com/ForrestErickson/PublicInvention_KiCad_Library)" by cloning from my repository.  Then we all can use that as the sub repository going forward.

RobertLRead, 2023-08-19:
This is done: https://github.com/PubInv/PublicInvention_KiCad_Library

You nasa-cog team has been invited to this repo.  I did it with a fork.

closed 2023-10-06 by LokiMetaSmith