PubInv/NASA-COG #145: Thermocouple sheaths should be grounded...

RobertLRead, 2023-08-21:
It is unclear to me where if this should be done to the frame (which is grounded to the high-power ground) or though a grounding why from a low voltage power supply carried through the DB25 cable to the fan controller.

Presumable, this grounding will decrease noise in the thermocouples.

LokiMetaSmith, 2023-08-21:
the shield wire for ground should be tied to the chassis and earth ground

LokiMetaSmith, 2023-09-28:
They make TC's that are bonded such that the TC wire is attached to the shield, and they make un grounded shielded TC's where the TC is isolated from the shield. 

They have different applications depending on how they're used.

ForrestErickson, 2023-10-04:
What is the status of this as of OEDCS Serial number 1 and 2?
Where can I find documentation on the shielding grounding connections?

ForrestErickson, 2023-10-05:
Please see the results of common mode (normal too) signal injected into thermocouple at this issue:
https://github.com/PubInv/NASA-COG/issues/273

With an oscilloscope and two probes the input of the thermocouple amplifiers should be characterized as a function of how the shield is grounded. 
Capture and journal measurements in an issue such as here.

Ground to minimize induced interference.

ForrestErickson, 2023-10-24:
Control V1.1 PCB has connector for grounding near the thermocouple amplifiers, J32, J33, and J43.

[image]

LokiMetaSmith, 2023-10-24:
These are tied to the ground plane of the low voltage DC bus and can potentially create an un-intended ground loop as the noise from the shielded TC's will be coupled to the sensitive DC low voltage connections. 

This needs to be evaluated and tested, where as the connections allow for inserting filtering capacitors and resistors, it is not clear if this is a practical solution. 

as per,              

 @LokiMetaSmith regarding, 
"I think we should plan on having pads for the ability to add passive filtering, DC-DC galvanic isolation, and support for TC shielding either into the main chassis or the analog front end." propose an exact circuit.

_Originally posted by @ForrestErickson in https://github.com/PubInv/NASA-COG/issues/273#issuecomment-1756608482_
            
We should leave this issue open until we have tested and evaluated the efficacy of the proposed solution