PubInv/NASA-COG #146: Should the GND of the 12V power supply be tied to the High-power and chassis ground?

RobertLRead, 2023-08-21:
At present I have wired the 12V GND out to the GND busbar, which may be thought of as a high-power GND. Lawrence points out this allows noise there to be communicated. It may be better to create a  low-power GND bus.

In particular, the DUE needs a low-power GND, but so does the blower and we need to ground the thermocouple sheaths which we have not yet done.

I thought it was better to have all GNDs tied together. I await a discuss in our Tuesday morning meeting to make a decision about this.

LokiMetaSmith, 2023-08-21:
treat the AC and LV_DC like isolated circuits with a single grounding point at the bus bar through the power supply. 

isolated DC power supplies can have a differential up to line voltage or higher, because the impedance between them is so high. This also brings up the question of adequate surge suppression and ESD protection, but that will likely need to wait until we have a design for a full control interface board

ForrestErickson, 2023-08-22:
Regarding, "... points out this allows noise there to be communicated."
Should be "... points out this allows interference there to be communicated."

Interference is the appropriate word because it keeps us thinking of how the "signals" of once circuit mix in with other circuits.  This mixing often well approximated as the voltages that appear as the signal currents go through common impedances (resistance if we are talking DC).   Wherever we can, we should reduce signal currents so as to reduce the amplitude of the voltage that mix signals on the common impedance.   
For circuits not at DC some of the circuit paths include the capacitive coupling and inductive coupling we must visualize because it is not happening due to resistance of the copper (or other conductors) of traces and wires.

**How to do this...**
By shaping the conductive paths so that signals do not share significant common paths of impedance we reduce the mixing of signals. This means that each signal needs to look like a circuit (a loop) which is independent of all other loops and the loops connect and one and only one point.    
This is not possible but to the extent we approximate it we reduce the mixing of signals.  
We reduce the mixing if we have  continuous (like the front panel) surface where all elements reference their ground, (common point of connection might be best called chassis ground).  Every circuit has in series a resistance and some times reactance that is as high as is compatible with the normal and desired operation of the circuit.

**An Insight**
For some circuits this is hard to imagine.  Lets take the Plant Power Supply, the programable +12V 60A circuit.  It is a "DC" circuit. However that is only an approximation. It is the filtered output with some remaining ripple of a switching power supply. It will have both a DC and and AC component on the long wires coming out of the supply (With in the supply there may also be large inductive sources of interference too.)  The AC component will inductively couple into other circuits. We minimize this coupling by minimizing the loop area.  We run the outgoing and incoming current through parallel conductors that are close together. Twisting the two conductors further helps because it changes one minimal loop in to many oppositely polarized loops that cancel out.   So if we were to find interference in some circuit that correlates the AC component of the Plant Power, that would be the first thing to consider.  In general minimizing the loop are of the victims circuit helps too. Again it is by "twisting" it.
Examining and understanding the geometry of the circuits (polluters and victims) is the critical approach.

Because the voltage on the 120 VAC heater circuits is the highest in our system we might get some capacitive coupling but I am still going to lean on inductive coupling as the most likely cause.  (Why, because the impedance is likely to be on the order of 120V/ 15 Amps = 8 Ohms which is less than he 337 impedance of free space) If interference were to be from the 120 VAC heater circuits ( the polluting circuit), understanding and minimizing the coupling of the inductive connection in to the victim circuit(s).  By comparison the plant supply 12volts/60amps is 0.2 Ohms so coupling is even more so possible by inductive coupling.  In the PIC processors with which I worked a typical GPIO had impedance of 25 to 50 ohms. I assume the Due will have the same order of magnitude. Once again lower than 337 Ohms of free space. So where ever we can introduce series resistance we will likely lower mixing of signals..

</end harangue>

RobertLRead, 2023-09-03:
Closing this issue as done.

closed 2023-09-03 by RobertLRead