PubInv/NASA-COG #151: Make Schematic Symbol and PCB Footprint, MAX31850

ForrestErickson, 2023-08-24:
Reference material: 
* https://learn.adafruit.com/connecting-the-max31855-thermocouple-amplifier-breakout-to-an-electric-imp
* https://www.adafruit.com/product/269

[image]

[image]

ForrestErickson, 2023-08-24:
@LokiMetaSmith
Please measure to see of the mounting holes on the MAX31850 are connected to the circuit ground.

ForrestErickson, 2023-08-24:
@LokiMetaSmith 
Please test fit all three types on this PDF.
[MAX31850_PCB20230824_0752.pdf](https://github.com/PubInv/NASA-COG/files/12429093/MAX31850_PCB20230824_0752.pdf)

Please leave feedback, photos, on the fit of the MAX31855 on this issue #151

LokiMetaSmith, 2023-10-06:
#247 is related to this issue, and can be closed when the PCB files are changed

ForrestErickson, 2023-10-24:
Done with V1.1

closed 2023-10-24 by ForrestErickson