PubInv/NASA-COG #160: Due_nRST2 signal, Do we want a push button?

ForrestErickson, 2023-08-24:
The Due has a pin name nRST2.

If the Ethernet shield obstructs the reset pin do we want a switch on this net on the Controller board?

RobertLRead, 2023-08-24:
> Assigned #160 <https://github.com/PubInv/NASA-COG/issues/160> to
> @RobertLRead <https://github.com/RobertLRead>.

Yes, we do -- but where is the pin?

ForrestErickson, 2023-08-25:
Here is the pin. I have placed a TAC switch on it.  Same part as the reset switch of the GPAD.
[image]

ForrestErickson, 2023-08-25:
Here is the PCB as I am Finishing tonight.
There are quite a few ERC errors to run down.....
[image]

[image]

ForrestErickson, 2023-08-29:
This has been incorporated into the schematic and PCB for V1.
Closing as done.

CC:
@LokiMetaSmith 
@RobertLRead

closed 2023-08-29 by ForrestErickson