PubInv/NASA-COG #171: Stray characters seemingly at random after every record in the UDP logs...

RobertLRead, 2023-08-28:
Oddly, there seems to be a stray character after each record in the log. I do not think I am adding it to the packet. In the example below, it is a double-quote charcter, but sometimes it changes. I wonder if this is in the logging code?
@gmulligan 

`
{ "TimeStamp": 1693240262,
"MachineState": "Warmup",
"TargetC": "28.00",
"HeaterC": "25.75",
"StackC": "25.25",
"GetterC": "25.50",
"StackV": "0.00",
"StackA": "0.00",
"StackOhms": "N/A",
"HeaterDutyCycle": "0.00",
"FanPWM": "0.38",
"FanRPM": "3079.00"
}"
{ "TimeStamp": 1693240262,
"MachineState": "Warmup",
"TargetC": "28.00",
"HeaterC": "25.50",
"StackC": "25.00",
"GetterC": "25.00",
"StackV": "0.00",
"StackA": "0.00",
"StackOhms": "N/A",
"HeaterDutyCycle": "0.00",
"FanPWM": "0.39",
"FanRPM": "3087.00"
}"
`

gmulligan, 2023-08-28:
I've fixed this on the actual server

closed 2023-08-28 by gmulligan

RobertLRead, 2023-08-28:
Awesome!