PubInv/NASA-COG #172: measurement logging

gmulligan, 2023-08-28:
In Cumulocity all "measurements" MUST be numbers...  They cannot be strings.  So we cannot include things like "inf" or "n/a" or on or off.

I've added code in the server that will change on or true to 1 and off or false to 0 and inf to 99999999 and n/a to -1, but the edge device should be fixed to send proper measurements.

If you want to send something that is a string, like the status of a device (on or off or heating or cooling) these are not measurements.  These are properties.

The difference is that measurements are time series and properties are not.

I don't remember off hand if properties have any historical memory.

RobertLRead, 2023-08-28:
I have replaced the "ohms" value with -1 when it is not a number or less than 0.0.

I don't know if that is the only instance of using a string in place of a number, but hopefully this has fixed the issue, so I am closing it.

gmulligan, 2023-08-28:
Currently Machine state is also a string.  I convert it from on off to 1 and 0

I think I'll change the server relay code to send that as both a measurement (o or 1) and as a property.

closed 2023-10-09 by RobertLRead