PubInv/NASA-COG #176: Unexplained full-heat event....

RobertLRead, 2023-08-29:
Today while I was working the heater turned on unexpectedly and unintentionally. I don't know.
I can see from the log at 2:21, everything was cold. I had a number of meetings and other work. I smelled something hot but thought it was the beans I was cooking.

When I worked on Bug # 175, which did not require entering start up mode, I saw the temperature was high, and unplugged the heater immediately.

At 8:57 the logs say the heater say:
`
{ "TimeStamp": 1693250471,
"MachineState": "Off",
"TargetC": "27.00",
"HeaterC": "26.75",
"StackC": "26.50",
"GetterC": "26.75",
"StackV": "7.06",
"StackA": "0.00",
"StackOhms": "-1.0",
"HeaterDutyCycle": "0.00",
"FanPWM": "0.00",
"FanRPM": "0.00"
}
{ "TimeStamp": 1693274247,
"MachineState": "Off",
"TargetC": "27.00",
"HeaterC": "828.75",
"StackC": "614.25",
"GetterC": "396.50",
"StackV": "0.00",
"StackA": "0.00",
"StackOhms": "0.00",
"HeaterDutyCycle": "0.00",
"FanPWM": "0.00",
"FanRPM": "0.00"
}
`

The fan was NOT operating during this time. The temperature across the stacks is more than 200 degrees; this quite possibly cracked the stack. At the moment it appears electrically okay.  Without the fan blowing, I don't know how hot the stack actually got. It has a large thermal mass, it may not have gotten as hot at the thermocouples around it suggest.

I assume that the SSR must have gotten stuck on, but I don't know how this would have happened. The system seems to have been in "off", not "warmup", based on the last log message.  I have no further explanation of this.

Possibly I have damaged the stack.

RobertLRead, 2023-08-29:
The system was turned on and connected to the power (and my main computer was monitoring the output.) I seem to recall that it was frozen when I came to it, but I cannot be certain.

The fact that no log files were produced during this period of time is consistent with the idea that the Arduino froze up. I have seen this behavior once before, but it was associate with an NPE that I thought I fixed.

It is possible that the Arduino foze up, but if that had happened I do not know why it would have happened with the heater on, when it was not a warm state. However, if the Arduino was locked up, the behavior of the GPIO pin might have been enough voltage to run on the SSR? This is the merest conjecture on my part.

If the Arduino is locking up, I don't know how to fix that. A watchdog could reset it, but it would have to be a different system which we do not currently have.

RobertLRead, 2023-08-29:
If the heater were turned on full blast, it would seem that it would have gotten hotter than 800 C with the fan not running. Could be that it was in some floating state, driven by an intermittent signal from the Arduino?

LokiMetaSmith, 2023-08-29:
Things to check

Inductive Noise - Do not wire power lines alongside the input lines of the SSR.

closed 2023-10-09 by RobertLRead