PubInv/NASA-COG #179: Order of Control V1 Assemblies, Loose Ends

ForrestErickson, 2023-08-29:
Email of 08/29/2023 12:55 AM EDT Lawrence Kincheloe

I have placed the order. 

> it took two hours to review, part select, fix random jlcpcb issues, and order. 
> 
> I'll make an issues later, but here's the burn down
> J10 needs to be a  B24B-PHDSS
> J13,J25,J11,J12 have wrong foot print, in the BOM it's listed as a 5.08mm footprint but the layout footprint is 2.54. I have selected a JST XH  which is 2.54 as a sub. I have the tooling to directly crimp connectors
> some versions of the BOM export tool truncate series values, such that C2-C14 is recorded as opposed to the expanded version, catching that and correcting that was a chore
> I had to export part positing  and BOM data from schematic layout, as those weren't in the repo at time of order
> I had to export gerbers, as they weren't exported in the repo at time of order
> I'll upload the files used to order with a date of ordering to the repository 
> 
> Delivery time of two boards is 2-10 days+ shipping 2-5 days

ForrestErickson, 2023-08-29:
Regarding, "Delivery time of two boards is 2-10 days+ shipping 2-5 days"
Clarification, this is delivery of two ASSEMBLIES  not raw PCBs.

ForrestErickson, 2023-08-29:
Regarding:
- [ ]   "J10 needs to be a B24B-PHDSS"  

The schematic shows part number as: **B24B-PHDSS**
So this is done??

FYI, the 3D model is only a header with pins not showing the plastic shroud of the actual part.

LokiMetaSmith, 2023-08-29:
when importing into JLC-PCB it didn't show up, although there was some
manual editing to get it in the form JLC-PCB would accept, so there's a
good chance it was an export error

ForrestErickson, 2023-08-29:
Regarding,

>  "J13,J25,J11,J12 have wrong foot print, in the BOM it's listed as a 5.08mm footprint but the layout footprint is 2.54. I have selected a JST XH which is 2.54 as a sub. I have the tooling to directly crimp connectors"

My bad. 
I had intended a screw terminal

I see: 
[image]

This is a SCREW TERMINAL part stocked at JLCPCB as C2915639.

> [image]
> This is a part I was and am considering for the Moonrat for 2.5Amps.

HOWEVER I did find a 2.54mm JLCPCB 
This part: **C918120** but apparently I did not update the part number in the schematic 
[image]

If you get a clarification question from JLCPCB and can change to **C918120** it might help you.
Or order something similar from Digikey / Mouser

I will update the schematic now.

ForrestErickson, 2023-08-29:
Regarding, 

> some versions of the BOM export tool truncate series values, such that C2-C14 is recorded as opposed to the expanded version, catching that and correcting that was a chore
> I had to export part [...]

I have a spread sheet, GPAD_V2_BOM.xls, for processing the KiCad BOM reports in the GPAD repository at:
https://github.com/PubInv/general-purpose-alarm-device/tree/main/Hardware/GeneralPurposeAlarmDevicePCB
**LINK:** [GPAD_V2_BOM.xls](https://github.com/PubInv/general-purpose-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/GPAD_V2_BOM.xls)  
Full instructions are built into the spread sheet to get from the KiCad CSV into a JLCPCB format file. It requires manual management of some ranges of copying from one tab to another.  

**CAUTION:** 
IIRC, I may have tested this with KiCad 7 BOM report data for Moonrat and there may be a different order to fields requiring a bit more under the hood work.

ForrestErickson, 2023-08-29:
Regarding,  "J10 needs to be a B24B-PHDSS"

Will this part 
**https://jlcpcb.com/partdetail/566831-B24B_XADSS_N_A_LF_SN/C545730**
work?
If so JLCPCB can assemble it.
Please advise.

ForrestErickson, 2023-08-30:
WARNING I am working on BOM now.

ForrestErickson, 2023-08-30:
JLCPCB BOM

| Comment                          	| Designator                                                                                                   	| Footprint                                                                                                  	| JLCPCB Part #  	|
|----------------------------------	|--------------------------------------------------------------------------------------------------------------	|------------------------------------------------------------------------------------------------------------	|----------------	|
| 39pF                             	| C1, C2, C3, C4, C5, C6, C7, C8, C9, C10, C11, C12, C13, C14, C15, C20, C21                                   	| Capacitor_SMD:C_0603_1608Metric_Pad1.08x0.95mm_HandSolder                                                  	| C107049        	|
| NCE01P30K                        	| Q2                                                                                                           	| Package_TO_SOT_SMD:TO-252-2                                                                                	| C115991        	|
| Conn_02x18_Odd_Even              	| J8                                                                                                           	| Connector_PinSocket_2.54mm:PinSocket_2x18_P2.54mm_Vertical                                                 	| C132131        	|
| SWITCH_TACTILE_SPST-NO_0.05A_24V 	| S1                                                                                                           	| PubInv_KiCad_PCB:Switch_Tactile_THT_6x6mm                                                                  	| C13828         	|
| DB9_Female_MountingHoles         	| J1                                                                                                           	| Connector_Dsub:DSUB-9_Female_Horizontal_P2.77x2.84mm_EdgePinOffset4.94mm_Housed_MountingHolesOffset7.48mm  	| C141882        	|
| 0.100_2Pin                       	| J22, J23                                                                                                     	| Connector_PinHeader_2.54mm:PinHeader_1x02_P2.54mm_Vertical                                                 	| C168673        	|
| DB25_Female_MountingHoles        	| J15                                                                                                          	| Connector_Dsub:DSUB-25_Female_Horizontal_P2.77x2.84mm_EdgePinOffset4.94mm_Housed_MountingHolesOffset7.48mm 	| C190083        	|
| 4M7                              	| R32                                                                                                          	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C23163         	|
| 1R0                              	| R1, R2, R3, R4, R5, R6, R7, R8, R13, R14, R15, R16, R31, R33, R48, R76, R79, R80, R82, R84, R88              	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C269434        	|
| DNI                              	| R17, R30, R34, R35, R43, R51, R52, R53, R55, R60, R63, R74, R75, R85, R86, R87, R89, R90                     	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C269434        	|
| 10K                              	| R39, R40, R41, R44, R46, R56, R61, R64, R65, R67, R68, R69, R70, R81                                         	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C269701        	|
| 1K                               	| R18, R19, R20, R21, R22, R23, R24, R25, R26, R27, R28, R29, R36, R37, R38, R42, R45, R47, R54, R57, R58, R62 	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C269704        	|
| 4K7                              	| R77, R78, R83                                                                                                	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                                                   	| C269713        	|
| Conn_01x08                       	| J2, J4, J5, J6                                                                                               	| Connector_PinSocket_2.54mm:PinSocket_1x08_P2.54mm_Vertical                                                 	| C2894931       	|
| Conn_01x08                       	| J3                                                                                                           	| Connector_PinHeader_2.54mm:PinHeader_1x08_P2.54mm_Vertical                                                 	| C2894931       	|
| Conn_01x10                       	| J7, J9, J14                                                                                                  	| Connector_PinSocket_2.54mm:PinSocket_1x10_P2.54mm_Vertical                                                 	| C2894933       	|
| Conn_01x05                       	| J17                                                                                                          	| Connector_PinHeader_2.54mm:PinHeader_1x05_P2.54mm_Vertical                                                 	| C2897368       	|
| Conn_01x04                       	| J16, J18, J24                                                                                                	| Connector_PinHeader_2.54mm:PinHeader_1x04_P2.54mm_Vertic [... cut ...]
| 1uF                              	| C16, C17, C18, C19                                                                                           	| Capacitor_SMD:C_0603_1608Metric_Pad1.08x0.95mm_HandSolder                                                  	| C91183         	|
| Screw_Terminal_01x02             	| J11, J12, J13, J25                                                                                           	| TerminalBlock_TE-Connectivity:TerminalBlock_TE_282834-2_1x02_P2.54mm_Horizontal                            	| C918120        	|

ForrestErickson, 2023-08-30:
Changes merged:
Make BOM for JLCPCB Parts #180

See the new files:
* Control_V1_BOM.xls
* Control_V1_BOM_JLCPCB_20230829_2137.csv

ForrestErickson, 2023-08-30:
FWIIW I spend 4.5 hours getting the JLCPCB part numbers into the schematic and then fighting spread sheet memories.
Must sleep now.

LokiMetaSmith, 2023-08-30:
bom for as-ordered parts

ForrestErickson, 2023-08-30:
Regarding, "bom for as-ordered parts"
Where?

LokiMetaSmith, 2023-08-30:
https://github.com/PubInv/NASA-COG/blob/develop/elec/Control%20Board/Control%20v1/Manufacturing%20Files/bom%20(3).xls

ForrestErickson, 2023-08-31:
Comparison of BOM as ordered and the BOM Lee generated after working on the schematic with JLCPCB parts as the main focus.

## Lawrence BOM as sent to JLCPCB
| Comment                                                                                         	| Designator                                                 	| Footprint                                                                       	| JLCPCB Part # 	|
|-------------------------------------------------------------------------------------------------	|------------------------------------------------------------	|---------------------------------------------------------------------------------	|---------------	|
| SWITCH_TACTILE_SPST-NO_0.05A_24V                                                                	| S1                                                         	| PubInv_KiCad_PCB:Switch_Tactile_THT_6x6mm                                       	| C13828        	|
| 39PF                                                                                            	| C3,C4,C5,C6,C7,C8,C9,C10,C11,C1,C2,C12,C13,C14,C15,C20,C21 	| Capacitor_SMD:C_0603_1608Metric_Pad1.08x0.95mm_HandSolder                       	| C107049       	|
| NCE01P30K                                                                                       	| Q2                                                         	| Package_TO_SOT_SMD:TO-252-2                                                     	| C115991       	|
| 1uF                                                                                             	| C16,C17,C18,C19                                            	| Capacitor_SMD:C_0603_1608Metric_Pad1.08x0.95mm_HandSolder                       	| C15849        	|
| Conn_02x12_Odd_Even                                                                             	| J10                                                        	| Connector_PinHeader_2.00mm:PinHeader_2x12_P2.00mm_Vertical                      	| C160327       	|
| HEADERVERT2POS2.54MM                                                                            	| J22,J23                                                    	| Connector_PinHeader_2.54mm:PinHeader_1x02_P2.54mm_Vertical                      	| C168673       	|
| 1R                                                                                              	| R49,R50,R59,R66,R73,R91,R92                                	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C22936        	|
| 4M7                                                                                             	| R32                                                        	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C23163        	|
| 1x2P20A-40℃~+105℃300VGreen12~30Straightpin5.08mm2.512Plugin,P=5.08mmScrewterminalROHS           	| J11,J12,J13,J25                                            	| TerminalBlock_TE-Connectivity:TerminalBlock_TE_282834-2_1x02_P2.54mm_Horizontal 	| C265283       	|
| Conn_01x04                                                                                      	| J16,J18,J20,J24                                            	| Connector_PinHeader_2.54mm:PinHeader_1x04_P2.54mm_Vertical                      	| C2691448      	|
| 1R0                                                                                             	| R31,R33,R48,R76,R79,R80,R82,R84,R88                        	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C269434       	|
| 10K                                                                                             	| R39,R40,R41,R44,R46,R56,R61,R64,R65,R67,R68,R69,R70,R81    	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C269701       	|
| 1K                                                                                              	| R42,R45,R47,R54,R57,R58,R62                                	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C269704       	|
| 4K7                                                                                             	| R77,R78,R83                                                	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C269713       	|
| 3AStraightSquarePins2.5mm8P6mm-40℃~+105℃3mm2.54mmPluginBlack1x8PPlugin,P=2.54mmPinHeadersROHS   	| J2,J4,J5,J6,J3                                             	| Connector_PinHeader_2.54mm:PinHeader_1x08_P2.54mm_Vertical                      	| C2894931      	|
| 3AStraightSquarePins2.5mm10P6mm-40℃~+105℃3mm2.54mmPluginBlack1x10PPlugin,P=2.54mmPinHeadersROHS 	| J7,J9,J14                                                  	| Connector_PinSocket_2.54mm:PinSocket_1x10_P2.54mm_Vertical                      	| C2894933      	|
| 39R                                                                                             	| R71,R72                                                    	| Resistor_SMD:R_0603_1608Metric_Pad0.98x0.95mm_HandSolder                        	| C325713       	|
| BSS138                                                                                          	| Q1,Q3,Q4,Q5,Q6,Q7,Q8,Q9,Q10                                	| Package_TO_SOT_SMD:SOT-23                                                       	| C400505       	|
[... cut ...]
| 1uF                              	| C16, C17, C18, C19                                                                                           	| Capacitor_SMD:C_0603_1608Metric_Pad1.08x0.95mm_HandSolder                                                  	| C91183         	|
| Screw_Terminal_01x02             	| J11, J12, J13, J25                                                                                           	| TerminalBlock_TE-Connectivity:TerminalBlock_TE_282834-2_1x02_P2.54mm_Horizontal                            	| C918120        	|

closed 2023-10-06 by LokiMetaSmith