PubInv/NASA-COG #181: Adding an automatic reset watchdog to guard against infinite loops

RobertLRead, 2023-08-30:
This explanation from the Geoff Mulligan

The WDT (watch dog timer/reset) is off by default.

if you want to use it you need to define

void watchdogSetup() {
watchdogEnable(millis);
}

Where millis is the number of milliseconds that the time will wait to be
fed.
As far as I can tell on the DUE you can only set this once!!!!  You
cannot change it after it is set.

To feed the watchdog you need to call:
     watchdogReset();

if you do not call the reset before the watchdog runs out of time the
processor will reset.
You must call watchdogReset();  using delay will not call it so a delay
of greater than millis will cause the processor/system to reset.

RobertLRead, 2023-08-30:
This note from Geoff:

If you want to check the reset reason here is some code...

  char *resetTypes[] = { "general", "backup", "watchdog", "software", "user" };
  uint32_t resetCause = rstc_get_reset_cause(RSTC) >> RSTC_SR_RSTTYP_Pos;

  Serial.print("ResetCause: ");
  Serial.println(resetTypes[resetCause]);

ForrestErickson, 2023-09-04:
More notes from Geoff
> On 08/29/2023 6:33 PM EDT Geoff Mulligan <[gmulligan@gmail.com](mailto:gmulligan@gmail.com)> wrote:
> The WDT (watch dog timer/reset) is off by default.
> if you want to use it you need to define
> void watchdogSetup() {
> watchdogEnable(millis);
> }
> Where millis is the number of milliseconds that the time will wait to be
> fed.
> As far as I can tell on the DUE you can only set this once!!!!  You
> cannot change it after it is set.
> 
> To feed the watchdog you need to call:
>     watchdogReset();
> 
> if you do not call the reset before the watchdog runs out of time the
> processor will reset.
> You must call watchdogReset();  using delay will not call it so a delay
> of greater than millis will cause the processor/system to reset.
> Geoff

From Lawrence
> On 08/30/2023 7:16 PM EDT Geoff Mulligan <gmulligan@gmail.com> wrote:
> If you want to check the reset reason here is some code...
> 
>   char *resetTypes[] = { "general", "backup", "watchdog", "software", "user" };
>   uint32_t resetCause = rstc_get_reset_cause(RSTC) >> RSTC_SR_RSTTYP_Pos;
> 
>   Serial.print("ResetCause: ");
>   Serial.println(resetTypes[resetCause]);
>

ForrestErickson, 2023-09-04:
From the Arduino reference:
https://www.arduino.cc/reference/en/libraries/watchdog/

> This library is compatible with the avr architecture so you should be able to use it on the following Arduino boards:
> 
> [Arduino Micro](https://store.arduino.cc/arduino-micro)
> [Arduino Leonardo](https://store.arduino.cc/arduino-leonardo-with-headers)
> [Arduino Mega](https://store.arduino.cc/arduino-mega-2560-rev3)
> [Arduino Nano](https://store.arduino.cc/arduino-nano)
> [Arduino Uno](https://store.arduino.cc/arduino-uno-rev3)
> [Arduino Yún](https://store.arduino.cc/arduino-yun-rev-2)
> 

Does not list Due

RobertLRead, 2023-09-04:
Interesting! Good catch! @Geoff Mulligan ***@***.***>

Please continue researching---possibly it is not available or no similar
function available on the Due.

ForrestErickson, 2023-09-05:
I have an elementary WDT working on  DUE. 
It times out when I hold a pin low and reports the reason for time out.

`/* WatchDog_Due_Eval.ino
 * 
 *  By:   (Forrest) Lee Erickson
 *  Date: 20230904
*/

/* Copyright (C) 2023 Forrest Lee Erickson and Public Invention

This program includes free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as
published by the Free Software Foundation, either version 3 of the
License, or (at your option) any later version.

See the GNU Affero General Public License for more details.
You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <https://www.gnu.org/licenses/>.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
*/

#define PROG_NAME "WatchDog_Due_Eval.ino"
#define VERSION "; Rev: 0.1"  //
#define BAUDRATE 115200
#define DEVICE_UNDER_TEST "Hardware: Due"

const long WATCH_DOG_TIME = 1500; // mS. Verified this has period of about 1.5 seconds
//20:25:05.732 -> Hardware: Due
//20:25:07.415 -> WatchDog_Due_Eval.ino; Rev: 0.1

//const int LED_BUILTIN = 13;
const int D7 = 7;     // An input Make low to make watchdog time out.
char *resetTypes[] = { "general", "backup", "watchdog", "software", "user" };  //Watch dog
//Functions

void watchdogSetup() {
  watchdogEnable(WATCH_DOG_TIME);
}

void checkD7() {
  while (!digitalRead(D7)); //It's a trap!
}//end checkD7

void setup() {
  // put your setup code here, to run once:
  Serial.begin(BAUDRATE);
  pinMode(D7, INPUT_PULLUP);
  digitalWrite(LED_BUILTIN, HIGH);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.print(PROG_NAME);
  Serial.println(VERSION);
  Serial.println(DEVICE_UNDER_TEST);

  //Print out the reset reason
  uint32_t resetCause = rstc_get_reset_cause(RSTC) >> RSTC_SR_RSTTYP_Pos;
  Serial.print("ResetCause: ");
  Serial.println(resetTypes[resetCause]);

  watchdogSetup();

  digitalWrite(LED_BUILTIN, LOW);   //At end of setup.
}//end setup()

void loop() {
  // put your main code here, to run repeatedly:

  //Do stuff
  checkD7();
  watchdogReset();

}//end loop()`

Screen shot:
[image]

ForrestErickson, 2023-09-05:
It took a couple of tries but I place code at: 
https://github.com/PubInv/NASA-COG/tree/develop/experimentation/WatchDog_Due_Eval

ForrestErickson, 2023-09-05:
I think it was mostly from code Lawrence posted.

The Arduino Library manager took me to code which worked for UNO and some others but not Due.

The code I wrote (From Lawrence's example) behaved as if the watchdog function was built in. No library required. I have not tried to figure out where it is to read further and will not tonight (punchy now).