PubInv/NASA-COG #188: E stop button to front control panel for ODECS #3-12

LokiMetaSmith, 2023-09-01:
add E-stop to the design requirement consideration for future ODECS systems. We already have plans for intrinsically safe shutdown on software and hardware failure, so this would be an external trigger for that system. 

This wouldn't disable power to the system, but it would disable output from the heaters and the programmable power supply. 

This would be separate from the control in the Due, but the Due should record the event when it happens and report that it's in E-stop mode.

ForrestErickson, 2023-09-02:
I think there is definitional work to be done.

A big red button should be triggering an event the goal of which, as best we and our customer can anticipate, is to quickly render the system off AND safe from **shock** and **fire hazards** and **other personal injury situations** characteristic of this oxygen generation and control system.

There are probably situations know to the current oxygen generation systems of which all of us should be learning.

## ODECS Serial Number 1 and 2 
For the ODECS Serial Number 1 and 2 the closest we have is the main power switch on the back and no formal failure effects mode analysis was undertaken when that switch was put into the design.

ForrestErickson, 2023-09-02:
## Emergency Stop
In factory assembly equipment I have seen the red mushroom button used to bring the system to a safe but not off state.
The big red button is easily accessible and intended to be used with out negative consequences so as not to discourage it's use.
[image]

## Emergency Off
Another kind of safety situation is like a crow bar across the electrical rail road tracks which is intended to for example blow a fuse. It has a higher level of safety but greater negative consequences.  Such a switch might require the operator take a more deliberate action to initiate the change of state such as lifting a switch cover.   Actuating the switch opens the AC mains (or blows a circuit breaker / fuse) .  It is a counter measure against a system out of control which the mushroom stop button has not corrected.

[image]

RobertLRead, 2023-09-03:
This is definitely a Task 2 issue. I don't want to spend time on this until we have delivered Task 1.

ForrestErickson, 2023-10-24:
Regarding, @RobertLRead statement, "This is definitely a Task 2 issue. "  
We have not front panel button on Serial Number 1 and 2. This cannot be a Task 2 firmware issue.  Move to later task(s).