PubInv/NASA-COG #192: If Ethernet card not present, we have a hard hang at start up....

RobertLRead, 2023-09-04:
At approximately line 209 of retreive_script_UDP_task.cpp, the call to "Udp.endPacket()" never returns if no etherent card is present.

Obviously, we don't create this task if we configure in src/main.cpp not to have an ethernet card (I use a #define on line 63).

However, I worry this is a risk if there is some other connectivity problem, or that it will be confusing and hard for a user to figure out. Can we make this more robust, so that if we timeout, or somehow survive this action, or never go down this path if we seem not to have a connectivity?

` Udp.beginPacket(timeServer, 123); //NTP requests are to port 123
    Serial.println("ZZZZZZ");
    Udp.write(ntpBuffer, NTP_PACKET_SIZE);
    Serial.println("MMMMMMM");
    Udp.endPacket();
    Serial.println("NNNNNNNN");
    delay(100);`

ForrestErickson, 2023-09-04:
I just found the following.
On my DUE with no ethernet card at boot the terminal reports:

> --- Terminal on COM11 | 115200 8-N-1
> --- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
> --- More details at https://bit.ly/pio-monitor-filters
> --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
> Starting Ox...        
> Create watchdog (todo)
> ReadTempsTask init
> Locating devices...Found 0 devices.
> Parasite power is: OFF
> Unable to find address for Device 0
> Unable to find address for Device 1
> Unable to find address for Device 2
> Device 0 Address: 0000000000000000
> Device 1 Address: 0000000000000000
> Device 2 Address: 0000000000000000
> Device 0 Resolution: 0
> Device 1 Resolution: 0
> Device 2 Resolution: 0
> Config of temperature sensors done
> CogTask init
> didn't set address
> didn't set address
> Manuf: UNKWN
> didn't set address
> Model:
> didn't set address
> VoltageSt:
> didn't set address
> Rev:
> didn't set address
> ManufDate:
> didn't set address
> Serial:
> didn't set address
> Country:
> didn't set address
> RateVoltage: 65535
> didn't set address
> RateCurrent: 65535
> MaxVoltage: 65535
> MaxCurrent: 65535
> didn't set address
> failed to turn it on
> didn't set address
> failed to set volts
> didn't set address
> failed to set current
> TempRefreshTask init
> starting
> The MAC address is:

ForrestErickson, 2023-09-04:
If I change the code to
#define ETHERNET_BOARD_PRESENT 0  //Make false

Then the terminal window shows

> Starting Ox...
> Create watchdog (todo)
> ReadTempsTask init
> Locating devices...Found 0 devices.
> Parasite power is: OFF
> Unable to find address for Device 0
> Unable to find address for Device 1
> Unable to find address for Device 2
> Device 0 Address: 0000000000000000
> Device 1 Address: 0000000000000000
> Device 2 Address: 0000000000000000
> Device 0 Resolution: 0
> Device 1 Resolution: 0
> Device 2 Resolution: 0
> Config of temperature sensors done
> CogTask init
> didn't set address
> didn't set address
> Manuf: UNKWN
> didn't set address
> Model:
> didn't set address
> VoltageSt:
> didn't set address
> Rev:
> didn't set address
> ManufDate:
> didn't set address
> Serial:
> didn't set address
> Country:
> didn't set address
> RateVoltage: 65535
> didn't set address
> RateCurrent: 65535
> MaxVoltage: 65535
> MaxCurrent: 65535
> didn't set address
> failed to turn it on
> didn't set address
> failed to set volts
> didn't set address
> failed to set current
> TempRefreshTask init
> Duty Cycle Setup
> DutyCycleTask init
> HeaterPIDTask init
> Added tasks
> Loop starting...
> Core::Run!
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 1
> Error: Could not read temperature data: 2
> Bad post_heater_C
> Bad post_getter_C
> Bad post_stack_C
> Running ReadTemps
> 
> Machine State: Off
> Target      C: 30.00
> Post Heater C: 0.00
> Post Getter C: 0.00
> Post Stack  C: 0.00
> Stack volts V: 0.00
> Stack amps  A: 0.00
> Stack ohms  O: 0.00
> Heater DC    : 0.0000
> Fan PWM      : 0.00
> Fan RPM      : 0.00
> Currrently Off. Enter a single 'w' to warmup: 
> didn't set address
> FAILED TO SET VOLTAGE!
> didn't set address
> didn't set address
> ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ
> Scheduler Still Alive, Number of Ticks:0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 1
> Error: Could not read temperature data: 2
> Bad post_heater_C
> Bad post_getter_C
> Bad post_stack_C
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 1
> Error: Could not read temperature data: 2
> Bad post_heater_C
> Bad post_getter_C
> Bad post_stack_C
> Running ReadTemps
> 
> Machine State: Off
> Target      C: 30.00
> Post Heater C: 0.00
> Post Getter C: 0.00
> Post Stack  C: 0.00
> Stack volts V: 0.00
> Stack amps  A: 0.00
> Stack ohms  O:  N/A
> Heater DC    : 0.0000
> Fan PWM      : 0.00
> Fan RPM      : 0.00
> Currrently Off. Enter a single 'w' to warmup:
> didn't set address
> FAILED TO SET VOLTAGE!
> didn't set address
> didn't set address
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 1
> Error: Could not read temperature data: 2
> Bad post_heater_C
> Bad post_getter_C
> Bad post_stack_C
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 0
> Error: Could not read temperature data: 1
> Error: Could not read temperature data: 2
> Bad post_heater_C
> Bad post_getter_C
> Bad post_stack_C
> Running ReadTemps
> 
> Machine State: Off
> Target      C: 30.00
> Post Heater C: 0.00
> Post Getter C: 0.00
> Post Stack  C: 0.00
> Stack volts V: 0.00
> Stack amps  A: 0.00
> Stack ohms  O:  N/A
> Heater DC    : 0.0000
> Fan PWM      : 0.00
> Fan RPM      : 0.00
> Currrently Off. Enter a single 'w' to warmup:
> didn't set address
> FAILED TO SET VOLTAGE!
> didn't set address
> didn't set address
> ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ
> Scheduler Still Alive, Number of Ticks:1
> Error: Could not read temperature data: 0

ForrestErickson, 2023-09-04:
Lee concludes that the ethernet code or the error checking of its return is brittle and that on an error (of initializing the ethernet card) the system became blocked.

RobertLRead, 2023-09-04:
Yes.  That is why I assigned #192 to @Geoff Mulligan ***@***.***>

ForrestErickson, 2023-09-18:
@RobertLRead ,
@gmulligan ,

The hard hang at start up has long since been cleared. 
Lee is closing.

closed 2023-09-18 by ForrestErickson