PubInv/NASA-COG #193: "make pio-run-due_ribbonfish" Fails On Lee's Windows 10

ForrestErickson, 2023-09-04:
Trying to build the firmware.

From the instructions at, https://github.com/PubInv/NASA-COG/tree/develop
Lee tried to run make "**pio-run-due_ribbonfish**"

Reference the instructions at:
[image]

Got this error:
[image]

ForrestErickson, 2023-09-04:
make : The term 'make' is not recognized as the name of a cmdlet, function, script
file, or operable program. Check the spelling of the name, or if a path was included,
verify that the path is correct and try again.
At line:1 char:1
+ make pio-run-due_ribbonfish
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (make:String) [], CommandNotFoundExcepti
   on
    + FullyQualifiedErrorId : CommandNotFoundException

ForrestErickson, 2023-09-04:
PIO is installed as best I can tell.

[image]

RobertLRead, 2023-09-04:
You probably can't do that in the powershell; it depends on upon
Linux-style commands.

Can you find a way to run a linux-like shell on Windows, such as "bash"?

The alternative is not use the "make" makefile, but to enter the commands
separately, which may work in a powershell:
The first is just intended to put you in the correct directory; the second
does a compilation and upload, and the third connects a serial "monitor"
(similar to xterm) to the Due:

cd firmware

pio run -e due_ribbonfish -t upload

pio device monitor --filter=direct --baud=115200

ForrestErickson, 2023-09-09:
I appear to have successfully built in Visual Studio Code with Platform IO. 
The program run in the target hardware and the terminal window shows activity.

[image]

Should the "How to run" be changed at the README.md at:
https://github.com/PubInv/NASA-COG/edit/develop/README.md
?

RobertLRead, 2023-11-17:
@ForrestErickson Can you please either make this change in the README.md or close this issue?