PubInv/NASA-COG #194: Build of Firmware Fails On Lee's Windows 10, Visual Studio and Platform IO,

ForrestErickson, 2023-09-04:
After failing at the CLI Lee tried to build in the Visual Studio Code IDE.

Compile produces an error:

> Compiling .pio\build\due_ribbonfish\liba70\task\stage2_heater_task.cpp.o
> In file included from lib\networking_task/Server.h:32:0,
>                  from .pio\libdeps\due_ribbonfish\Ethernet\src/Ethernet.h:53,
>                  from lib\task\retrieve_script_UDP_task.cpp:35:
> lib\networking_task/mongoose.h:42:2: error: #error "MG_ARCH is not specified and we couldn't guess it."
>  #error "MG_ARCH is not specified and we couldn't guess it." 
>   ^~~~~
> lib\networking_task/mongoose.h:59:10: fatal error: mongoose_custom.h: No such file or directory
> 

Screen shot full IDE
[image]

RobertLRead, 2023-09-04:
This is confusing to me. I thought I had removed those directories. Give me
10 minutes.

RobertLRead, 2023-09-04:
I have removed it; do a pull and try it again.

ForrestErickson, 2023-09-04:
I have pulled.

I appear to have built and loaded into the target the firmware.

[image]

ForrestErickson, 2023-09-04:
After pressing reset I can confirm that the Cog task is running.

[image]

ForrestErickson, 2023-09-04:
The last thing printed into the terminal is "The MAC address is: "
Then the Due becomes unresponsive.

RobertLRead, 2023-09-04:
@Geoff Mulligan ***@***.***> yes, I was afraid of this----there is
a hang if you attempt to configure the Ethernet card when it is not
present. This is issue #192 I created and assigned to Geoff.

You can get around it by changing "src/main.cpp" near line 63 to NOT define
the constant as positive, make it be like this:

#define ETHERNET_BOARD_PRESENT 0

.. instead of 1.  Then it will not hang (at least not there.)

ForrestErickson, 2023-09-09:
I have installed the Seeed Studio Ethernet card W5200

The code now builds and runs.
[image]

ForrestErickson, 2023-09-09:
After connecting the ethernet cable and connecting to my network the traffic on the terminal in Visual Code Studio stopped.

[image]

I pressed reset. Traffic resumed and I noticed the MAC address go buy so here it is:

> failed to set current
> TempRefreshTask init
> starting
> The MAC addressUnix time = 1694270399
> The UTC time is 14:39:59
> Duty Cycle Setup
> DutyCycleTask init
> HeaterPIDTask init
> Added tasks
> Loop starting...
> Core::Run!
> .....

ForrestErickson, 2023-09-09:
Terminal traffice is steady so at 10:42 I disconnected the Ethernet

Serial port traffice continued.
[image]

ForrestErickson, 2023-09-18:
While I do not remember what changes I made, I have been able to make the project for some time now.

Closing.

closed 2023-09-18 by ForrestErickson