PubInv/NASA-COG #204: Add Machine configuration script to Makefile

LokiMetaSmith, 2023-09-05:
Machine configuration script should allow for:

- Defining pinout in one location
- Define number and enablment of heaters, fans, and accessories
- Define machine specific information such as MAC address, unit name, serial number, etc

gmulligan, 2023-09-05:
I'm not sure what of this I'm supposed to do...  The MAC address is hard-coded into the Ethernet Board.

RobertLRead, 2023-09-05:
I don't think this should be assigned to you.

gmulligan, 2023-09-05:
OOPS my mistake!  The MAC address is assigned on the ethernet board by the code and this should be separated into a config file and not buried inside the udp task library.

BUT we need to figure out how to assign mac addresses!!!!!

gmulligan, 2023-09-05:
also Lawrence is right, the pinouts for the ethernet and CS should be also pulled into a separate config.h or some such.  But I don't think I'm qualified to make these changes.

ForrestErickson, 2023-09-05:
Regarding, "The MAC address is hard-coded into the Ethernet Board."
and then "OOPS my mistake! The MAC address is assigned on the ethernet board by the code"

It is easy to get this confused.  I would have never guessed that the shield did not come with a MAC assigned by the manufacturer.

Where is there documentation on the ethernet shield to which we can refer?

RobertLRead, 2023-09-05:
I think @gmulligan  should answer this.

ForrestErickson, 2023-09-05:
If the manufacturer does not provide a MAC address and if you want to sell a product and insure you do not use an address that belongs to some other entity,  you must purchase MAC addresses. The IEE sells them. If I remember correctly this is going to be the smallest increment, MAC Address Block Small (MA-S),
https://standards.ieee.org/products-programs/regauth/oui36/

The cost is  $830 for a range of 4096 addresses (the 12 LSB are assigned by the manufacturer) or about $0.20 each.

ForrestErickson, 2023-09-18:
Lee is not sure he understands / agrees with this request.
Lee's difficulty is because of several things. One is unfamiliarity with make files and how they might be used to configure firmware.
Second, at least the item, Serial Number, in the list above is not appropriate for user entry at compile time . 
A serial number should be added into an assembled unit way after code is compiled and released and then loaded into a unit. This means serial number gets stored in the Due Flash.

I suggest this issue should have a more descriptive name reflective of the scope to which a Makefile is appropriate.

Regarding, "Defining pinout in one location"  I would think this definition needs to be very visible to everyone using the code so that we can be sure we understand the pinout as we do the things we need to do, Hardware design, system test, firmware development and so on. Somewhere where a Control Shift F (for project find) will return useful results.

LokiMetaSmith, 2023-09-18:
This has been addressed with calling separate machine hal defines, for example, class Stage2HAL : public MachineHAL { as defined in stage2_hal.h

in platformio.ini the build flag -D STAGE2_HEATER=1 selects relevant definitions

closed 2023-10-09 by RobertLRead