PubInv/NASA-COG #207: Test the ability to survive broken ethernet connection without a hang

RobertLRead, 2023-09-05:


ForrestErickson, 2023-09-09:
On 20230908 Lee received the **seeed studio Ethernet Shield** he ordered.
Is there some simple firmware with which I can verify the card connectivity?

ForrestErickson, 2023-09-09:
Found a seeed studio tutorial at:
https://wiki.seeedstudio.com/Ethernet_Shield_V2.0/

Installing the library with Library Manager
[image]

ForrestErickson, 2023-09-09:
Loading example file
[image]

Fails to compile.
Document failure at Seeed Studio repository for the library at:
https://github.com/Seeed-Studio/Ethernet_Shield_W5200/issues/14

ForrestErickson, 2023-09-18:
### Testing Notes
Testing on MOM with [c79bc3c](https://github.com/PubInv/NASA-COG/commit/c79bc3c42ce0ccab385441132d64aecffc58d61a)).

1.  Loaded firmware into DUT and booted. Watched traffic on serial port using the Arduino Serial Monitor. 
With Ethernet connected the DUT ran for over a minute with apparent normal operation. It never reported an IP address. 
Lee interupted the Ethernet cable for a minuet or so and the DUT behavior did not change. 
Lee reconnected the Ethernet cable and the DUT behavior did not change. 

2. Capture from Start up
 Same as before but make sure to capture the start up.
MAC Address: FE:ED:16:6B:72:67
13:29:26.932 -> Unix time = 1695058165
13:29:26.932 -> The UTC time is 17:29:25
13:29:27.168 -> Network started
Interrupted Ethernet at about 13:33:??
DUT worked as normal.
Reconnect Ethernet at about 13:37:00
DUT worked as normal.
Check log and at 13:39 and the log only showed the one MAC address from about 13:29 above.

Conclusion. The DUT does not "hang" with a broken ethernet connection.

 Closing issue. Make new issue if this problem reemerges.

ForrestErickson, 2023-09-18:
Closing issue. Make new issue if this problem reemerges.

closed 2023-09-18 by ForrestErickson