PubInv/NASA-COG #209: Modify system to provide time ramp of amperage

RobertLRead, 2023-09-05:
On Sept. 5th we learned that too much power too quickly to a stack is a dangerous condition. Furhermore, 60 amps may be to high. 

We need a more sophisticated control algorithm for amperage. This was a major event and possibly a major problem.

LokiMetaSmith, 2023-09-05:
This should be in therm of Wattage, as we need to measure the added thermal load onto the system. 

We need to be aware that as the stack self heats, we won't see that reflected on the TC's upstream and down stream, until the heat has migrated across the stack, through the air (which is our thermal transfer medium) to the TC. There is additional loss involved with the external insulation as well, so it's not a perfect 1 to 1

One thing we can do is that we can measure the resistance of the stack, based on the current draw to give an estimate of the heat being generated. This won't account for hotspots however because a single hot high current conducting spot can emulate a broad power transfer across the entire stack, with the difference being that the single spot cannot effectively transfer heat as well as the entire stack

With that in mind, I recommend the following. 

We need to better model the proper operation of the stack, there's a couple of ways to do this.

Build a table/database of all possible proper stack states and how they can transfer from one to another. Out of bound tranistions can trigger an error

build a heat model of the stack, with representative hidden variables that can be modeled and estimated from first principals and real time sensor data as well as energy inputs

(unlikely but also a fix) add sense lines to each wafer stack to monitor voltage potential across each stack for real time monitoring of electrical reluctance 

Run an AUTO PID control loop for the entire stack, treat the power supply as a PID controller with the thermocouple tied to the reluctance to electrical conductivity across the stack.

RobertLRead, 2023-11-17:
@LokiMetaSmith I'm closing this issue. You make some suggestions here which you may wish to move to a different, more comprehenisve issue.

closed 2023-11-17 by RobertLRead