PubInv/NASA-COG #211: Missing resistors in bom

LokiMetaSmith, 2023-09-06:
[image]

1R0 and 1R some populated and some didn't.

Will need to apply jumpers where nessisary

ForrestErickson, 2023-09-07:
Back on about Aug 30 or so, I put 1K resistors on these pins during my clean up of the BOM for JLCPCB parts.

[image]

LokiMetaSmith, 2023-10-06:
These were jumpered, leaving issue for control v2

ForrestErickson, 2023-10-06:
FYI, Work I did on the schematic (BOM) after the V1 assemblies was ordered filled in these parts and more.
[image]
Specifically I used 1K so that even if the jumpers on the Adafruit assemblies were not cut correctly there would be no smoke let out of anything.

RobertLRead, 2023-11-17:
I assume this is closable.

closed 2023-11-17 by RobertLRead