PubInv/NASA-COG #212: J10 mirrored on pcb

LokiMetaSmith, 2023-09-06:
Schematic layout is correct, but pcb is mirrored such that pin 1 and pin 2 need to swap.

Can be recovered, but power cable will need to be modified to work with pcb boards

ForrestErickson, 2023-09-06:
Ah connectors....

Please describe how this is determined?

If you could unsolder the connector and solder to bottom that would fix the problem local to the assembly and allow a cable to be used with future boards where this is fixed.

LokiMetaSmith, 2023-10-06:
J10 was also flipped as well as mirrored, the cable was simply repinned and the modified end was made obvious with sharpie as to the correct orientation. 

leavig this open until changed for Control v2

ForrestErickson, 2023-10-06:
Regarding, "J10 was also flipped as well as mirrored, "  

Gathering information.

JST connector specification detail.
[image]

Our PCB footprint.
[image]

It appears our PCB footprint must mirror to become something like this.
[image]

LokiMetaSmith, 2023-10-06:
That is correct.

RobertLRead, 2023-11-17:
@ForrestErickson  is this closable?