PubInv/NASA-COG #217: Thermocouple reads fail 1 - in a hundred times; or maybe more under some conditions...

RobertLRead, 2023-09-07:
The thermocouples are returning 10/1673 bad reads. This was in a simple case of warmup with a low duty cycle. I believe I have seen other circumstances when this is more common. This is dealt with in software be catching very low temperature readings, so it is not a problem at this low failure rate but it may be indicative of a greater failure.

Machine State: Warmup
Target      C: 55.50
Post Heater C: 58.00
Post Getter C: 27.00
Post Stack  C: 26.75
Stack volts V: 12.07
Stack amps  A: 0.07
Stack ohms  O: 172.43
Heater DC    : 0.1020
Fan PWM      : 0.60
Fan RPM      : 4262.00
Target Temp : 55.50
Good Temp Reads:1673
Bad  Temp Reads:10

RobertLRead, 2023-09-08:
After running over night, there is strong circumstantial evidence that the bad reads occur when the SSR is active and do not occur when the SSR is not active.

With the Due up for about 16 hours, we obtain:
SSR OFF - 0 / 1300 bad/ good reads
SSR ON  - 1  / 1407 bad / good reads
SSR ON - 3  /  1566 bad / good reads
SSR ON - 9 / 2139  bad / good reads
The the SSR was turned off with an emergency stop and we went to:
SSR OFF- 9 / 14987 bad / good reads

In this experiment, the highest Duty cycle reached was 27% on the SSR.
An obvious action is to change the wire dress (which is very haphazard at present) and see if we can induce the problem more commonly. This would be evidence that we can use directly to solve the problem.

RobertLRead, 2023-09-08:
There is now compelling evidence that our bad temperature reads are caused by some kind of EMI.

When placing the thermocouples amplifiers closer the SSR as shown in the photo below, the bad read read increase drastically. Although still usable, we see 112/782 failures when the SSR is largely off. When the DC equal 50%, this increases drastically, from about 16% of failures to as much as 33%.

With this system in "off" mode (which also turns off the fan) , the rate (when lower) drops to 2%.

Note the comment above, when this the circuit board holding the amplifiers is about 6" higher, (on the top rail of the case), the fail rate drops to about 1/1000.

RobertLRead, 2023-09-08:
@ForrestErickson What do you suggest that I do about this in terms of wire dressing or managment inside the case?

I assume this is interfering with the digital part of the amplifier, not the voltage in the wires itself.

ForrestErickson, 2023-09-09:
### VERY Good detective work here

> There is now compelling evidence that our bad temperature reads are caused by some kind of EMI.
>
> When placing the thermocouples amplifiers closer the SSR as shown in the photo below, the bad read [...]

### Conjecture 
The AC wires at the relay have current passing through and forms a dipole loop which magnetically threads through other loops.

### Test Conjecture, Make worse yet
You could perhaps make coupling and therefor the problem  worse by spreading the two wires with the AC power to increase the loop area.  Placing the thermocouple stuff near one wire with in the loop may make it worse too.

### Multipole as a possible fix, 
Turn that single loop formed by the two white wires (or perhaps a white and grey wire) into a much higher order multipole which largely cancels out by twisting the two wires as depicted in this photo.
[image]

ForrestErickson, 2023-09-14:
## Testing on Maryville Mockup (Due with W5200 ethernet shield and one MAX31850 )
Had failures getting readings and made notes in email "Notes on Max31850 Familiarization" of  9/13/2023 10:11 PM
[Notes on Max31850 Familiarization Printout20230913.pdf](https://github.com/PubInv/NASA-COG/files/12607228/Notes.on.Max31850.Familiarization.Printout20230913.pdf)
Finally got one thermocouple reading after 10:00 PM and captured some data.

### Hardware Description
Due 
W5200 ethernet shield 
One MAX31850 on solderless breadboard
The thermocouple is unshielded. It is in a 5 to 6 " diameter loop laying on cables from two wall warts, and some oscilloscope probes.  Connected but not energized is a bench supply and the blower. 

Firmware: "MAX31850_Temperature_example"  Based on MAX31850_Temperature but with different BAUD rate and add 
>   delay(200);
>   Serial.print("Starting: ");
>   Serial.println(PROGRAM_NAME);
> 

**Capture log with Tera Term from 2023-09-13 22:29:52.729 to 2023-09-14 04:59:59.157 so about 6.5 hours.**

* First reading
> [2023-09-13 22:29:52.729] ******************************************
> [2023-09-13 22:29:52.740] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:29:52.740]   Chip = MAX31850
> [2023-09-13 22:29:52.740]   Data = 1 68 1 30 16 F0 FF FF FF EC  CRC=EC
> [2023-09-13 22:29:52.740]   Address = 0x0
> [2023-09-13 22:29:52.740]   Temperature = 22.50 Celsius, 72.50 Fahrenheit
> [2023-09-13 22:29:52.740] ******************************************

* A Bad reading at [2023-09-13 22:29:52.756]. It was preceded and followed by neraly normal readings of 22.xx degrees HOWEVER the reading immediatly before was 26.50 which is at least 3 degrees higher.

> [2023-09-13 22:29:52.740] ******************************************
> [2023-09-13 22:29:52.740] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:29:52.740]   Chip = MAX31850
> [2023-09-13 22:29:52.740]   Data = 1 6C 1 30 16 F0 FF FF FF F9  CRC=F9
> [2023-09-13 22:29:52.740]   Address = 0x0
> [2023-09-13 22:29:52.740]   Temperature = 22.75 Celsius, 72.95 Fahrenheit
> [2023-09-13 22:29:52.740] ******************************************
> [2023-09-13 22:29:52.740] 
> [2023-09-13 22:29:52.740] No more addresses.
> [2023-09-13 22:29:52.740] 
> [2023-09-13 22:29:52.740] ******************************************
> [2023-09-13 22:29:52.740] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:29:52.740]   Chip = MAX31850
> [2023-09-13 22:29:52.740]   Data = 1 A8 1 30 16 F0 FF FF FF 37  CRC=37
> [2023-09-13 22:29:52.740]   Address = 0x0
> [2023-09-13 22:29:52.740]   Temperature = 26.50 Celsius, 79.70 Fahrenheit
> [2023-09-13 22:29:52.740] ******************************************
> [2023-09-13 22:29:52.756] 
> [2023-09-13 22:29:52.756] No more addresses.
> [2023-09-13 22:29:52.756] 
> [2023-09-13 22:29:52.756] ******************************************
> [2023-09-13 22:29:52.756] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:29:52.756]   Chip = MAX31850
> [2023-09-13 22:29:52.756]   Data = 1 FC 7F 30 16 F0 FF FF FF 93  CRC=93
> [2023-09-13 22:29:52.756]   Address = 0x0
> [2023-09-13 22:29:52.756]   Temperature = 2047.75 Celsius, 3717.95 Fahrenheit
> [2023-09-13 22:29:52.756] ******************************************
> [2023-09-13 22:29:52.756] 
> [2023-09-13 22:29:52.756] No more addresses.
> [2023-09-13 22:29:52.756] 
> [2023-09-13 22:29:52.756] ******************************************
> [2023-09-13 22:29:52.756] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:29:52.756]   Chip = MAX31850
> [2023-09-13 22:29:52.756]   Data = 1 64 1 30 16 F0 FF FF FF D3  CRC=D3
> [2023-09-13 22:29:52.756]   Address = 0x0
> [2023-09-13 22:29:52.756]   Temperature = 22.25 Celsius, 72.05 Fahrenheit
> [2023-09-13 22:29:52.756] ******************************************
>

ForrestErickson, 2023-09-14:
## More analysis of log file Log_MAX31850_Temperature_example_20230913_2229.txt

I see the program start message is not at the top of file.  I may have not cleared the buffer before starting the log properly.
The first two reads show "FAULT" and the third read is wild high. Only the forth read is sensable at 22.25C

> [2023-09-13 22:30:02.121] Starting: MAX31850_Temperature_example
> [2023-09-13 22:30:02.125] ******************************************
> [2023-09-13 22:30:02.137] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:30:02.142]   Chip = MAX31850
> [2023-09-13 22:30:03.149]   Data = 1 FD 7F 31 16 F0 FF FF FF E7  CRC=E7
> [2023-09-13 22:30:03.157]   Address = 0x0
> [2023-09-13 22:30:03.157] **FAULT!**
> [2023-09-13 22:30:03.157] ******************************************
> [2023-09-13 22:30:03.161] 
> [2023-09-13 22:30:03.161] No more addresses.
> [2023-09-13 22:30:03.165] 
> [2023-09-13 22:30:03.404] ******************************************
> [2023-09-13 22:30:03.420] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:30:03.420]   Chip = MAX31850
> [2023-09-13 22:30:04.431]   Data = 1 FD 7F 31 16 F0 FF FF FF E7  CRC=E7
> [2023-09-13 22:30:04.440]   Address = 0x0
> [2023-09-13 22:30:04.440] **FAULT!**
> [2023-09-13 22:30:04.440] ******************************************
> [2023-09-13 22:30:04.444] 
> [2023-09-13 22:30:04.444] No more addresses.
> [2023-09-13 22:30:04.447] 
> [2023-09-13 22:30:04.686] ******************************************
> [2023-09-13 22:30:04.702] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:30:04.702]   Chip = MAX31850
> [2023-09-13 22:30:05.714]   Data = 1 FC 7F 30 16 F0 FF FF FF 93  CRC=93
> [2023-09-13 22:30:05.722]   Address = 0x0
> [2023-09-13 22:30:05.722]   Temperature = 2047.75 Celsius, 3717.95 Fahrenheit
> [2023-09-13 22:30:05.726] ******************************************
> [2023-09-13 22:30:05.730] 
> [2023-09-13 22:30:05.730] No more addresses.
> [2023-09-13 22:30:05.734] 
> [2023-09-13 22:30:05.972] ******************************************
> [2023-09-13 22:30:05.984] ROM = 3B 38 B7 94 C 4C 6C B4
> [2023-09-13 22:30:05.988]   Chip = MAX31850
> [2023-09-13 22:30:07.000]   Data = 1 64 1 30 16 F0 FF FF FF D3  CRC=D3
> [2023-09-13 22:30:07.004]   Address = 0x0
> [2023-09-13 22:30:07.008]   Temperature = 22.25 Celsius, 72.05 Fahrenheit
> [2023-09-13 22:30:07.012] ******************************************
> [2023-09-13 22:30:07.012] 
> [2023-09-13 22:30:07.016] No more addresses.
> [2023-09-13 22:30:07.016] 

Perhaps there should be some time allotted after a reset before reading.  But why?  It is not as if the power just came on.

ForrestErickson, 2023-09-14:
## Instrumenting the MAX31850_Temperature_exampl.ino
Set D6 to go high at start of conversion, Low after end.

From code

>  ds.reset();
>   ds.select(addr);
>   digitalWrite(TRIGGER_PIN, HIGH);  //High at start of conversion
>   ds.write(0x44, 1);        // start conversion, with parasite power on at the end
>   digitalWrite(TRIGGER_PIN, LOW);  //Low at end of conversion
> 
>   delay(1000);     // maybe 750ms is enough, maybe not
>   // we might do a ds.depower() here, but the reset will take care of it.

### Examine Waveforms
[image]
**CH1 D5 data, CH4 D6 trigger**

Same thing but zoom on the section where D6 is high.  
[image]
**CH1 D5 data, CH4 D6 trigger**

So the data before might be the reset and select address.
Move the trigger like this:

>   digitalWrite(TRIGGER_PIN, HIGH);  //High at start of conversion
>   ds.reset();
>   ds.select(addr);
> //  digitalWrite(TRIGGER_PIN, HIGH);  //High at start of conversion
>   ds.write(0x44, 1);        // start conversion, with parasite power on at the end
>   digitalWrite(TRIGGER_PIN, LOW);  //Low at end of conversion
> 

The resulting waveform
[image]

**Guess:** 
The data before the reset must be the reading (and printing) of the previous measurement.

ForrestErickson, 2023-09-14:
## MOM with two MAX31850
I have connected a second thermocouple to the second amplifier board.
When I read two MAX31850 at once I get a fault.
[image]

### After messing only slightly with the wires I am reading both.
I rewired the solderless bread board only slightly so that I could easily read one TC at at time.
[image]

Read TC0 only
[image]

Read TC1 only
[image]

ForrestErickson, 2023-09-14:
Measure 3V3 supply during data conversion.

On TC1 measure the 3V3 and see a slight droop with a duration of about 72mSec.

[image]

ForrestErickson, 2023-09-14:
Notes on error rate.
Ran from 8:17 to 9:50 the MAX31850_Temperature_example; Rev 0.2 and had zero "FAULT", zero "CRC is not valid!"
Manual spot check shows all temperatures were reasonable. There appear to be zero errors out of something like 4K readings.

ForrestErickson, 2023-09-14:
Make new Arduino sketch to graph the temperature readings
MAX31850_Blower.ino

Some data:
[image]

The data started out nice and smooth but then got wild.  It got wild as I was typing and doing other computer work moving my arms around the test station.  Finally I held the coil of TH wire to TC0 and got a run of poor measurements.

ForrestErickson, 2023-09-14:
## Experiment with FAN on.

Same as above but disconnect the PWM signal that was low and holding the fan off.

[image]

A bit later.
[image]

ForrestErickson, 2023-09-14:
## Examining TC Amplifier Input

Signals on TC0 + and - thermo couple input. Blue (?Cyan?) is the + and magenta is the -.
[image]

Grip TC0 coil with hand
[image]
And another
[image]

ForrestErickson, 2023-09-14:
Examine the relation between the Data, the D6 (ds.reset...) and the thermocouple signals

[image]

The oscilloscope waveform above corresponds to the left most (first) event on this graph.
[image]

ForrestErickson, 2023-09-14:
## BAD Alligator Clip
The root cause of the erratic readings above was that one of the alligator clips on TC0 was failing.  When I was gripping the loop of wire it was flexing that connection which finally failed.  Once I repaired it the readings are stable.

Here is a data set where TC1 was in ice water (note that the measurement is in error) and one I pinched in my fingers during the run and released. 
[image]

I was using a modified example code from the Dallas 1 Wire library which made much faster readings. Note that it is also reading only 9 bit resolution.

Here is the code which made this plot:
[Dallas_Tester.zip](https://github.com/PubInv/NASA-COG/files/12613321/Dallas_Tester.zip)

ForrestErickson, 2023-09-14:
@LokiMetaSmith 
@gmulligan 

FYI, please see the the information just above.
I now have stable temperature readings. (Bad alligator clip)

ForrestErickson, 2023-09-15:
@RobertLRead 
@LokiMetaSmith 

From the data sheet for the one wire thermocouple amplifier, MAX31850-MAX31851.pdf we have the following:

> The scratchpad memory contains the 2-byte temperature register that stores the cold-junction-compensated thermocouple temperature data. A second 2-byte register  stores the local cold-junction temperature. Both of these 
> registers also store fault data for open thermocouple as well as shorts to supply and ground

Perhaps we need to look into the libraries to see if we can interpret these three different results and report them. 

Same for the MAX31855 parts.

LokiMetaSmith, 2023-09-15:
The libraries already support exposing the error conditions, there's even a
ToDo, somewhere to add it in when we go to read the TC value to make sure
there's no errors as well.

This would be under the, "Good first Issue" category as the problem and
solution are well defined

ForrestErickson, 2023-09-18:
@RobertLRead ,
@LokiMetaSmith 

## Retrospection morning of 20230918
### Difficulty during wiring the MOM
Though it did not get into the reports above I spoke about the mystery of how wiring the D0 line from one MAX313850 module to the other on the MOM station did not work in one configuration, About a 1.5 inch daisy chain but worked when I relocated the pull up resistor and made a "Y" like connection  to the two MAX31850 modules.  And that I had not tried to duplicate the failed wiring configuration once the two devices began to deliver data.  Today, 20230918 I moved a wire to return to a daisy chain like configuration and the temperature reads fine.  
**Conclusion**, As I was messing with the wiring of the D0 my arm was also moving the alligator clips to the thermocouples and changing the connection now know to be mechanically intermittent / marginal.

### Adding 0.1uF to the Thermocouple
Not captured in graphs above was the experiment where on TC0 I added 0.1uF across the thermocouple. Wat I saw was that the oscilloscope waveform during an abnormal read was not as big and the measured temperature rather than often going to about 2200 C went more often to something like 600C.  where I am guessing the 2200 like reading was the full scale. 
This was a clue that the connection to the thermocouple was intermittent though I (Lee) did not recognized it at that time. It was during a meeting with Robert, Lawrence and Lee that we realized we were using different libraries (Was this on 20230914?).  My notes above fail to show when or even that I added the 0.1uF though I think it was just before the meeting but perhaps after.
Regardless adding the 0.1uF often reduced the severity of the intermittent connection.  
**Conclusion,** The Atafruit evaluation boards likely do not have a 0.1uF at the TC input.

### Readings of FAULT and DEVICE_DISCONNECTED_C
In the text file Log_MAX31850_Temperature_example_20230913_2229.txt the word FAULT shows up and if our reports had been decoding the three possible errors the MAX31850 to use and we understood them we might have better insight. This FAULT was reported by the code using the Adafruit library.

Using the Dallas library code, here is a time plot where I disconnected from TC0 the + then the - and then from TC1 the + and then the - leads at the alligator clips to the copper wires.
[image]
Remember that TC0 has 0.1uF capacitor. TC1 does not. 
NOTE how the legend on the graph changed 
[image]

The code had the following code path in yellow background.
[image]

I believe both of these libraries were trying to tell something useful about the fault conditions and I was learning only slowly.

From the Dallas library keywords file at: https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/keywords.txt
We see the following related to faults 
```
#######################################
# Constants (LITERAL1)
#######################################

DEVICE_DISCONNECTED_C	LITERAL1
DEVICE_DISCONNECTED_F	LITERAL1
DEVICE_DISCONNECTED_RAW	LITERAL1
DEVICE_FAULT_OPEN_C	LITERAL1
DEVICE_FAULT_OPEN_F	LITERAL1
DEVICE_FAULT_OPEN_RAW	LITERAL1
DEVICE_FAULT_SHORTGND_C	LITERAL1
DEVICE_FAULT_SHORTGND_F	LITERAL1
DEVICE_FAULT_SHORTGND_RAW	LITERAL1
DEVICE_FAULT_SHORTVDD_C	LITERAL1
DEVICE_FAULT_SHORTVDD_F	LITERAL1
DEVICE_FAULT_SHORTVDD_RAW	LITERAL1
```

It appears there are four fault types, Disconnected, Open, Short to gnd and short to VDD which can be reported.

**Conclusion.**  Our production code should be alerting us distinctly to all fault conditions distinctly for each thermocouple location.

RobertLRead, 2023-09-18:
Interestingly, last night I coded an EmergencyShutdown based on a
thermocouple disconnect.  It seems to work. (see below).

HOWEVER, on an actual disconnect I did NOT see the value read -127.00 as it
should. When I pull out the thermocouple, it read -0.19 degrees C.

I suspect "DISCONNECT" means "from the chip", not the thermocouple.  The
code below does noting for the SPI thermocouples; possible @Lawrence
Kincheloe ***@***.***> can work on that. Possibly you can improve
the code here as well, Lee.

float ReadTempsTask::evaluateThermocoupleRead(int
idx,CriticalErrorCondition ec,int &rv) {
  float temp = _temperatureSensors[0].GetTemperature(idx);

#ifdef USE_MAX31850_THERMOCOUPLES
  // we'd like to use the corret sentinels, but they don't seem to work...
  //  if (temp == DEVICE_DISCONNECTED_C) {
  if (temp < 0.0) {
    if (!getConfig()->errors[ec].fault_present) {
      getConfig()->errors[ec].fault_present = true;
      getConfig()->errors[ec].begin_condition_ms = millis();
      Serial.println("THERMOCOUPLE DISCONNECTED!!!!!!!!!");
    }
  } else {
    getConfig()->errors[ec].fault_present = false;
  }

  return temp;
#elif
  // probably the SPI based MAX31855_THERMOCOUPLES
#endif
}

RobertLRead, 2023-09-18:
[image: image.png][image: image.png]

Lee, not to belabor the point, but it DOES have a 0.1uF, putting an extra
0.1uF should further reduce noise immunity at the proportional decrease in
rapid temperature rise rate sensitivity.

" The fault-detection circuit tests for shorted connections to VDD or GND
on the T+ and T- inputs, as well as looking for an open thermocouple
condition."
BUT,

/*!
   * @brief attempt to determine if the device at the given address is
connected
   * to the bus
   * @param deviceAddress Address to search for
   * @return Returns true if the device is connected
   */
  bool isConnected(uint8_t *);

Connected, being an overloaded term, in this context means finding the temp
sensor on the 1-wire bus.

ForrestErickson, 2023-09-18:
Regarding, "Lee, not to belabor the point, but it DOES have a 0.1uF, putting an extra
0.1uF should further reduce noise immunity at the proportional decrease in
rapid temperature rise rate sensitivity."
I pulled an evaluation board free from the Solderless Breadboard and confirmed there is a capacitor across the + and - inputs.

ForrestErickson, 2023-09-19:
Testing update
The code 

> 19:40:29.807 -> Starting OEDCS; Rev: 0.3.4
> 19:40:29.807 -> Build: Sep 18 2023 19:39:30  

ran for some hours. 
At 20230919 2:41 reports only one error on TC0 out of 5060 total.
[image]

AND this was after Lee had been connecting and disconnecting the thermocouple at an alligator clip during the operation.

LokiMetaSmith, 2023-10-06:
There are only a few things we can do to fix this in future iterations. 

- Provide grounding lugs or accommodations for shielded, and ungrounded tip thermocouples
- add 60Hz specific filters on TC input lines
- Use DC-DC isolators to remove ground loops going back to the electric grid
- Use twisted TC wire for long runs
- Adding filters decrease TC response rates, so that will need careful consideration in conjunction with the TC reported response rate requirement
- Wire management to minimize EMI from both magnetic and static sources. 
- Identify a dummy noise injection method of reproducing the problem reliably