PubInv/NASA-COG #218: SPI interface differences between Uno and Due

LokiMetaSmith, 2023-09-08:
Notice that on the Uno have pins 11,12,13 tied to the ICSP programming header and in the Due 11,12,13 **are not associated with SPI**

This took some digging, but the Ethernet shield uses the ICSP header, and not pins 11,12,13 (they aren't even connected) 
which matches with why the ethernet shield, "just works". Also Pins 4, 10, 52 are unique 

The repercussions of this are subtitle. For one, neither of the din rail break out boards also bring out SPI pins from the ICSP, using the display isn't  

Uno
[image]

DUE
[image]

here's a better reference for 
[image]

source: https://forum.arduino.cc/t/due-pinout-diagram/129258

and the full pinout
https://content.arduino.cc/assets/A000056-full-pinout.pdf

RobertLRead, 2023-09-08:
"using the display isn't "   - what? your comment seems garbled.

Nonetheless this is great research.

LokiMetaSmith, 2023-09-08:
I've corrected the post above. 

Additional information. A work around is to use software SPI, with the cavate that the bit rate will be slower, but for this application that maybe more than acceptable.

ForrestErickson, 2023-09-08:
Regarding, 
"This took some digging, but the Ethernet shield uses the ICSP header, and not pins 11,12,13 (they aren't even connected)
which matches with why the ethernet shield, "just works". Also Pins 4, 10, 52 are unique"

Regarding "pins 11,12,13 " do you mean on your Ethernet shield?  The Due?

To which exact make and model Ethernet  shield are you referring?

I have ordered, but not yet received, seeed studio Ethernet Shield.
https://www.amazon.com/dp/B00W0HZVMS?psc=1&ref=ppx_yo2ov_dt_b_product_details
Mine is Arriving September 15 - September 26.

ForrestErickson, 2023-09-08:
Regarding SPI to thermocouples, 
For reference, here is the PCB routing
[image]

With a flying wire you could connect the SPI clock from the test point to the header pin to get SPI to the D13 if that helps.  The KiCad "measurement tool" on this screen shot evokes such a connection.
[image]

ForrestErickson, 2023-09-08:
Similar for MISO
[image]

ForrestErickson, 2023-09-08:
MOSI and pin D11
[image]

ForrestErickson, 2023-09-08:
Is it right to assume the nCS could be D10 for the Ethernet?
[image]

ForrestErickson, 2023-09-08:
@LokiMetaSmith 
Regarding the issue name, "SPI interface differences between Uno and Due"
Does this make a problem, with for example the Ethernet shield? And if so please describe the problem. Edit the original details may help, may be the best place to do so.

ForrestErickson, 2023-09-08:
Regarding, " A work around is to use software SPI,"  Do you mean " A work around is to use software SPI, for the Ethernet card"?

LokiMetaSmith, 2023-09-08:
> Regarding, "This took some digging, but the Ethernet shield uses the ICSP header, and not pins 11,12,13 (they aren't even connected) which matches with why the ethernet shield, "just works". Also [...]

The Ethernet shield v2 from Seeed Studio also do not use pin 11,12,13
[image]

LokiMetaSmith, 2023-09-08:
> @LokiMetaSmith Regarding, " A work around is to use software SPI,"
> 
> Do you mean " A work around is to use software SPI, for the Ethernet card"?

A work around to not having access to the ICSP header is to use software SPI. This is presently only the case on the Stage 2 heater controller

LokiMetaSmith, 2023-09-08:
> @LokiMetaSmith
> Regarding the issue name, "SPI interface differences between Uno and Due"
> Does this make a problem, with for example the Ethernet shield? And if so please describe the problem. Edit the original details may help, may be the best place to do so.

This can cause confusion and misunderstanding, as a quick search for "Due and SPI" will attest. In my own case, the poor documentation around it, caused quite a bit of confusion because the ethernet shield does use SPI, but it's not clear where those pins are defined, and the naive assumption is that the Due is pin to pin compatible with the Uno, which is only partly true

LokiMetaSmith, 2023-09-09:
[image]
This was fixed in the stage 2 heater with a ribbon cable and direct wire of d4 to d4 and d10 to d10

LokiMetaSmith, 2023-10-09:
closing issue

closed 2023-10-09 by LokiMetaSmith